cortex-lab/phy #810: Did not filter data before sorting, now trying to get waveforms

hamdanspam, 2019-01-02:
Hello, I have extracted waveforms from my raw data based on best channel and best units, but I realize now that the filtering Kilosort does when it sorts spikes is not saved in the raw.dat file! Therefore my waveforms are unfiltered and basically unusable.

Is there a way to filter my raw.dat file so that I can recover my waveforms the way they are shown in the template-gui?

Many thanks!

hamdanspam, 2019-01-04:
I tried to make a filter that would not time-shift my data at all, here is the code I came up with that would filter my raw.dat post-hoc.

```
fid = fopen('raw.dat', 'w'); % open .dat file for writing

amplifier_data_filtered = bandpass(amplifier_data' ,[350, 4900],...
        frequency_parameters.amplifier_sample_rate,...
        'ImpulseResponse', 'fir'); % fir does not shift peak of trough positions
amplifier_data_filtered = amplifier_data_filtered';
fwrite(fid, amplifier_data_filtered(:),'int16');
```

Then, I was able to run `getWaveForms.m` from the `spikes` repository on my data just fine.

closed 2019-01-04 by hamdanspam

SamuelHosovsky, 2019-01-06:
In your other comment you mentioned, you were not able to run getWaveForms function at all. Here you state you were eventually able to run it - mind sharing your version of the getWaveForms function? I'm currently stuck on the memmapfile issues sh9346c@greenwich.ac.uk

hamdanspam, 2019-01-08:
Hey Samuel! Thanks for your message, the code snippet I use to run getWaveForms.m can be found [here](https://github.com/cortex-lab/spikes/issues/9). Hope this help.