cortex-lab/phy #811: Error: No such command "template-gui"

ScottHarris17, 2019-01-09:
Hello,

I'm new to python am trying to use phy on windows 10. I've followed all installation instructions but can't seem to get the gui to launch. I've downloaded the [small test data set](https://phy-contrib.readthedocs.io/en/latest/template-gui/#installation) but when I try to launch the gui from the command line using 'phy template-gui params.py' I get 'Error: No such command "template-gui".'

I tried following the solution from this similar [thread ](https://github.com/kwikteam/phy/issues/775) with no luck 

Anything helps.

Thanks!

Gery111, 2019-01-10:
I guess you are not install it correctly

aliddell, 2019-03-21:
I had to reinstall Phy and ran into this same problem. I used the environment.yml from phy-contrib since the one from phy doesn't have everything.

First trying `python -c "import phy, phycontrib"`, I got "No module named 'PyQt4.QtCore'" so I went [here](https://www.lfd.uci.edu/~gohlke/pythonlibs/#pyqt4) and got PyQt4. Since a `python --version` gives Python 3.5.5 and I'm on 64-bit Windows, I downloaded `PyQt4‑4.11.4‑cp35‑cp35m‑win_amd64.whl`. In the `phy` conda environment, I navigated to my Downloads folder and did `pip install PyQt4‑4.11.4‑cp35‑cp35m‑win_amd64.whl`, so now I have PyQt4.

Next `python -c "import phy, phycontrib"` gave me "No module 'tqdm'" so I did `pip install tqdm`. After this, no message.

Phy still crashes, but for a different reason.

rossant, 2019-06-13:
should be fixed in the dev branch with update to PyQt5

closed 2019-06-13 by rossant