cortex-lab/phy #814: WaveformView does not show spike waveforms

wehrlab, 2019-02-07:
Hi,
I am using Kilosort to cluster tetrode data that's collected in Open Ephys. I have installed and used phy before and had not experienced this issue. It looks like the waveform view is showing flat lines, instead of spike waveform. It does not appear to be a scaling issue, since changing the scale does not fix it. I'm wondering if there is a variable that needs to be adjusted in my configuration file during clustering or if there are some libraries that I'm missing. Thanks. 
[image]

rossant, 2019-02-12:
Could you try to rename, in your home directory, your `~/.phy` to `~/.phy~` (there might be abnormal scaling parameters in the autogenerated config files)? Same for the `.phy` subfolder in your data repository.

adehad, 2019-05-05:
@wehrlab I had a similar abnormal scaling issue after moving to the `gloo` branch. I used the `Ctrl+Up` shortcut until I could see my waveforms.

closed 2019-05-27 by rossant