cortex-lab/phy #819: Dev branch installation help? Invalid function arguments & [33m & [0m.

adehad, 2019-04-27:
I seem to have some issues with my installation version of the dev branch, on Windows 10, Python 3.5.6.

In addition to this seemingly incorrect newline and/or end of line character, I am unable to toggle the mean waveform or templates view.

```
[33m18:35:27 [W] actions:171          Invalid function arguments: expecting 0 but got 1[0m
[33m18:35:28 [W] actions:171          Invalid function arguments: expecting 0 but got 1[0m
```
I feel that this might be related to my installation?
This is my conda environment file (we still use `kwik-gui` occaisionally hence the `klusta` dependencies):
```
name: phy-dev
channels:
  - defaults
dependencies:
  - blas=1.0=mkl
  - ca-certificates=2018.03.07=0
  - certifi=2018.8.24=py35_1
  - cycler=0.10.0=py35hcc71164_0
  - cython=0.28.5=py35h6538335_0
  - git=2.18.0=h6bb4b03_0
  - h5py=2.8.0=py35h3bdd7fb_2
  - hdf5=1.10.2=hac2f561_1
  - icc_rt=2017.0.4=h97af966_0
  - intel-openmp=2018.0.3=0
  - jpeg=8d=vc14_2
  - libpng=1.6.34=h79bbb47_0
  - libtiff=4.0.6=vc14_2
  - mkl=2018.0.3=1
  - numpy=1.11.3=py35h53ece5f_10
  - numpy-base=1.11.3=py35h8128ebf_10
  - openssl=1.0.2p=hfa6e2cd_0
  - pip=10.0.1=py35_0
  - pyparsing=2.2.0=py35_1
  - pyqt=4.11.4=py35_7
  - python=3.5.6=he025d50_0
  - python-dateutil=2.7.3=py35_0
  - pytz=2018.5=py35_0
  - qt=4.8.7=vc14_9
  - scipy=1.1.0=py35hc28095f_0
  - setuptools=40.2.0=py35_0
  - sip=4.18=py35_0
  - six=1.11.0=py35_1
  - tk=8.5.18=vc14_0
  - vc=14=h0510ff6_3
  - vs2015_runtime=14.0.25123=3
  - wheel=0.31.1=py35_0
  - wincertstore=0.2=py35hfebbdb8_0
  - zlib=1.2.11=h8395fce_2
  - pip:
    - chardet==3.0.4
    - click==6.7
    - decorator==4.3.0
    - idna==2.7
    - ipython-genutils==0.2.0
    - joblib==0.12.4
    - kiwisolver==1.0.1
    - klusta==3.0.16
    - klustakwik2==0.2.6
    - matplotlib==3.0.3
    - git+https://github.com/kwikteam/phy.git@dev
    - git+https://github.com/kwikteam/phy-contrib.git@dev 
    - pyopengl==3.1.0
    - pyopengl-accelerate==3.1.0
    - pyqt5==5.12.1
    - pyqt5-sip==4.19.15
    - pyqtwebengine==5.12.1
    - requests==2.19.1
    - tqdm==4.26.0
    - traitlets==4.3.2
    - urllib3==1.23
    - vispy==0.5.3
prefix: phy-dev
```

I don't feel like the `klusta` packages are affecting this installation, as this is a cloned environment of my original `phy` environment which works fine. I did need to update `matplotlib`, but cannot determine what version I would require from the provided environment file.

Would appreciate any insight.
Cheers!

rossant, 2019-04-30:
I think this bug is fixed in the upcoming dev version (currently in the gloo branch)

adehad, 2019-05-05:
Can confirm that moving over to the `gloo` branch worked! 
I used the following in my conda environment:
```
pip install git+https://github.com/kwikteam/phy.git@gloo --upgrade
pip install git+https://github.com/kwikteam/phy-contrib.git@gloo --upgrade
pip uninstall PyOpenGL-accelerate
```

I had to remove `PyOpenGL-accelerate` due to an error, shown below for completeness:
```
(phy-dev) C:\MY_DATA_FOLDER>phy template-gui params.py
c:\conda\miniconda3\envs\phy-dev\lib\site-packages\OpenGL\GL\VERSION\GL_2_0.py:374: FutureWarning: comparison to `None` will result in an elementwise object comparison in the future.
  baseOperation(program, index, length, namelen, size, gl_type, name)
Traceback (most recent call last):
  File "c:\conda\miniconda3\envs\phy-dev\lib\site-packages\phy\plot\base.py", line 438, in paintGL
    visual.on_draw()
  File "c:\conda\miniconda3\envs\phy-dev\lib\site-packages\phy\plot\base.py", line 82, in on_draw
    self.program.draw(self.gl_primitive_type)
  File "c:\conda\miniconda3\envs\phy-dev\lib\site-packages\phy\plot\gloo\program.py", line 597, in draw
    self.activate()
  File "c:\conda\miniconda3\envs\phy-dev\lib\site-packages\phy\plot\gloo\globject.py", line 87, in activate
    self._activate()
  File "c:\conda\miniconda3\envs\phy-dev\lib\site-packages\phy\plot\gloo\program.py", line 383, in _activate
    attribute.activate()
  File "c:\conda\miniconda3\envs\phy-dev\lib\site-packages\phy\plot\gloo\globject.py", line 87, in activate
    self._activate()
  File "c:\conda\miniconda3\envs\phy-dev\lib\site-packages\phy\plot\gloo\variable.py", line 367, in _activate
    self.data.activate()
  File "c:\conda\miniconda3\envs\phy-dev\lib\site-packages\phy\plot\gloo\globject.py", line 95, in activate
    self._update()
  File "c:\conda\miniconda3\envs\phy-dev\lib\site-packages\phy\plot\gloo\buffer.py", line 90, in _update
    gl.glBufferSubData(self.target, offset, nbytes, data)
  File "latebind.pyx", line 44, in OpenGL_accelerate.latebind.Curry.__call__ (src\latebind.c:1201)
  File "c:\conda\miniconda3\envs\phy-dev\lib\site-packages\OpenGL\GL\VERSION\GL_1_5.py", line 122, in glBufferSubData
    data = ArrayDatatype.asArray( data )
  File "arraydatatype.pyx", line 174, in OpenGL_accelerate.arraydatatype.ArrayDatatype.asArray (src\arraydatatype.c:4221)
  File "arraydatatype.pyx", line 57, in OpenGL_accelerate.arraydatatype.HandlerRegistry.c_lookup (src\arraydatatype.c:2230)
TypeError: No array-type handler for type <class 'phy.plot.gloo.buffer.VertexBuffer'> (value: VertexBuffer([ 0,  0,  0, ...,  0, 64, 64], dtype=) registered
```

closed 2019-05-05 by adehad