cortex-lab/phy #825: Suggestion: use `quality` instead if `group`

alejoe91, 2019-05-28:
Hi @rossant 

As you probably know by now we have set up a framework for spike sorting with several sorters (https://github.com/SpikeInterface). In this framework we can `export_to_phy` template-gui format the output of all sorters.

Internally, we use `group` to indicate the `channel_group`, which can be loaded with standard .prb files. However, in phy the `group` metadat indicates whether a cell is good, noise, unsorted, or mua.

I think that `quality` would be a better naming for it. Also I suggest keeping `group` to indicate the channel group. If no group is indicated, then all group will be 0. This comes very handy, for example, for tetrode recordings, where the template_ind and feature_ind are slit by channel group.

What do you think?

PS: is the original kwikteam/phy moved here officially? Was the `gloo` branch merged into `dev`?

Cheers,
Alessio

rossant, 2019-05-28:
Hi
The repo was moved here indeed, and gloo was merged into dev. I'll give more details soon!
I agree that the current naming is not ideal and your proposition is sound. However, I worry about compatibility as changing the names now might confuse previous users. Especially because the name `group` would mean different things according to the version of phy, and similarly for `quality` which is the name used in the "wizard" for sorting. So unfortunately I'm not certain we're able to change the naming at the moment.

alejoe91, 2019-05-28:
Ok!
So at the moment we use a custom metadata (`ch_group`). Maybe that could be used? 
Also, when splitting and merging the user-defined metadata go lost. 
Maybe a quick fix could be:
- when merging the metadata is inherited from the cluster with more spikes
- when splitting the metadata of the children is inherited from the parent

Could something like this be easily implemented? :)

Cheers
Alessio

rossant, 2019-05-28:
Ah, that's a bug, thanks. I'll fix it!

closed 2019-05-30 by rossant