cortex-lab/phy #829: Memmap cannot use existing file handles

JoseGuzman, 2019-05-31:
I obtained Phy from the latest dev (2.0a0), and after clustering with spyking-circus, I get the error below. 

I would so much appreciate some help!

```python
(python3) segundo.martinez@pcl-imba-3:spyking-circus$ phy template-gui phy/params.py 
Traceback (most recent call last):
  File "/home/segundo.martinez/miniconda3/envs/python3/bin/phy", line 11, in <module>
    load_entry_point('phy', 'console_scripts', 'phy')()
  File "/home/segundo.martinez/miniconda3/envs/python3/lib/python3.5/site-packages/click/core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "/home/segundo.martinez/miniconda3/envs/python3/lib/python3.5/site-packages/click/core.py", line 697, in main
    rv = self.invoke(ctx)
  File "/home/segundo.martinez/miniconda3/envs/python3/lib/python3.5/site-packages/click/core.py", line 1066, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/segundo.martinez/miniconda3/envs/python3/lib/python3.5/site-packages/click/core.py", line 895, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/segundo.martinez/miniconda3/envs/python3/lib/python3.5/site-packages/click/core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "/home/segundo.martinez/miniconda3/envs/python3/lib/python3.5/site-packages/click/decorators.py", line 17, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/segundo.martinez/git/phy/phy/apps/__init__.py", line 112, in cli_template_gui
    template_gui(params_path)
  File "/home/segundo.martinez/git/phy/phy/apps/template/gui.py", line 829, in template_gui
    controller = TemplateController(**params)
  File "/home/segundo.martinez/git/phy/phy/apps/template/gui.py", line 87, in __init__
    self.model = TemplateModel(dat_path, **kwargs)
  File "/home/segundo.martinez/miniconda3/envs/python3/lib/python3.5/site-packages/phylib/io/model.py", line 193, in __init__
    self._load_data()
  File "/home/segundo.martinez/miniconda3/envs/python3/lib/python3.5/site-packages/phylib/io/model.py", line 217, in _load_data
    self.spike_samples = self._load_spike_samples()
  File "/home/segundo.martinez/miniconda3/envs/python3/lib/python3.5/site-packages/phylib/io/model.py", line 431, in _load_spike_samples
    return self._read_array(path)
  File "/home/segundo.martinez/miniconda3/envs/python3/lib/python3.5/site-packages/phylib/io/model.py", line 316, in _read_array
    return read_array(path).squeeze()
  File "/home/segundo.martinez/miniconda3/envs/python3/lib/python3.5/site-packages/phylib/io/model.py", line 43, in read_array
    return np.load(path, mmap_mode='r')
  File "/home/segundo.martinez/miniconda3/envs/python3/lib/python3.5/site-packages/numpy/lib/npyio.py", line 430, in load
    return format.open_memmap(file, mode=mmap_mode)
  File "/home/segundo.martinez/miniconda3/envs/python3/lib/python3.5/site-packages/numpy/lib/format.py", line 754, in open_memmap
    raise ValueError("Filename must be a string.  Memmap cannot use"
ValueError: Filename must be a string.  Memmap cannot use existing file handles.
```

rossant, 2019-05-31:
are you able to do some quick and dirty debugging for me? Open `/home/segundo.martinez/miniconda3/envs/python3/lib/python3.5/site-packages/phylib/io/model.py` in a text editor, go to line 43, and replace the line by:

```
print(path)
return np.load(str(path), mmap_mode='r')
```
try again and send me the output. Thanks!

JoseGuzman, 2019-05-31:
Sure, whatever is necessary!

> try again and send me the output. Thanks!

Exactly the same output!
```python
(python3) segundo.martinez@pcl-imba-3:spyking-circus$ phy template-gui phy/params.py 
/home/segundo.martinez/SiliconProbes/spyking-circus/phy/spike_times.npy
Traceback (most recent call last):
  File "/home/segundo.martinez/miniconda3/envs/python3/bin/phy", line 11, in <module>
    load_entry_point('phy', 'console_scripts', 'phy')()
  File "/home/segundo.martinez/miniconda3/envs/python3/lib/python3.5/site-packages/click/core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "/home/segundo.martinez/miniconda3/envs/python3/lib/python3.5/site-packages/click/core.py", line 697, in main
    rv = self.invoke(ctx)
  File "/home/segundo.martinez/miniconda3/envs/python3/lib/python3.5/site-packages/click/core.py", line 1066, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/segundo.martinez/miniconda3/envs/python3/lib/python3.5/site-packages/click/core.py", line 895, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/segundo.martinez/miniconda3/envs/python3/lib/python3.5/site-packages/click/core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "/home/segundo.martinez/miniconda3/envs/python3/lib/python3.5/site-packages/click/decorators.py", line 17, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/segundo.martinez/git/phy/phy/apps/__init__.py", line 112, in cli_template_gui
    template_gui(params_path)
  File "/home/segundo.martinez/git/phy/phy/apps/template/gui.py", line 829, in template_gui
    controller = TemplateController(**params)
  File "/home/segundo.martinez/git/phy/phy/apps/template/gui.py", line 87, in __init__
    self.model = TemplateModel(dat_path, **kwargs)
  File "/home/segundo.martinez/miniconda3/envs/python3/lib/python3.5/site-packages/phylib/io/model.py", line 194, in __init__
    self._load_data()
  File "/home/segundo.martinez/miniconda3/envs/python3/lib/python3.5/site-packages/phylib/io/model.py", line 218, in _load_data
    self.spike_samples = self._load_spike_samples()
  File "/home/segundo.martinez/miniconda3/envs/python3/lib/python3.5/site-packages/phylib/io/model.py", line 432, in _load_spike_samples
    return self._read_array(path)
  File "/home/segundo.martinez/miniconda3/envs/python3/lib/python3.5/site-packages/phylib/io/model.py", line 317, in _read_array
    return read_array(path).squeeze()
  File "/home/segundo.martinez/miniconda3/envs/python3/lib/python3.5/site-packages/phylib/io/model.py", line 44, in read_array
    return np.load(path, mmap_mode='r')
  File "/home/segundo.martinez/miniconda3/envs/python3/lib/python3.5/site-packages/numpy/lib/npyio.py", line 430, in load
    return format.open_memmap(file, mode=mmap_mode)
  File "/home/segundo.martinez/miniconda3/envs/python3/lib/python3.5/site-packages/numpy/lib/format.py", line 754, in open_memmap
    raise ValueError("Filename must be a string.  Memmap cannot use"
ValueError: Filename must be a string.  Memmap cannot use existing file handles.
(python3) segundo.martinez@pcl-imba-3:spyking-circus$ 
```

rossant, 2019-05-31:
It looks like it's still the old cached version that was used. Could you install the GitHub version of phylib and do the change there?

JoseGuzman, 2019-06-02:
Hi, 

It works only when I changed the line in model.py
```python
#return np.load(path, nmap_mode='r')
return np.load(path)
```

Would you please tell me how to do update the newest GitHub version? What I did is:
```console
foo@server:~$ conda activate python3
foo@server:~$ git clone https://github.com/cortex-lab/phy.git
foo@server:~$ cd phy
foo@server:~$ git checkout dev
foo@server:~$ pip install -e .
```
Which returns:
```console
  Found existing installation: phy 1.0.9
    Uninstalling phy-1.0.9:
      Successfully uninstalled phy-1.0.9
  Running setup.py develop for phy
Successfully installed phy
```

rossant, 2019-06-02:
You should also use the master version of phylib:

```
git clone https://github.com/cortex-lab/phylib.git
pip install -e .
```

and do the proposed changed there.

That being said, if it works without the memmap and you if don't get memory issues, I guess you're fine... It's unclear what I'm unable to reproduce this error though.

JoseGuzman, 2019-06-03:
Hi @rossant ,

Thank you very much for your help!. It took me the whole weekend to play around this

I've just installed *phylib* and my conda environment moves from python3.5 to python3.7. The memory map issue is gone after updating to phylib. To run *phy* required only few updates (pyOpenGL, colorcet). 

Thank you very much for your great assistance!

closed 2019-06-03 by JoseGuzman