cortex-lab/phy #830: Installation Issue w/ pip

jaredluckas, 2019-05-31:
Hi,

I'm attempting to work with Phy, but I keep running into an issue with the installation. I'm getting this error:

>C:\Users\Jared\Documents\phy-master>conda env create -n phy
Warning: you have pip-installed dependencies in your environment file, but you do not list pip itself as one of your conda dependencies.  Conda may not use the correct pip to install your packages, and they may end up in the wrong place.  Please add an explicit pip dependency.  I'm adding one for you, but still nagging you.
>Collecting package metadata: failed

Does anyone know how to fix this? I'm extremely new to this so a short explanation of what this all means might be helpful.

yagui, 2019-06-06:
what conda version are you using? anaconda? miniconda?
with newer version I would just use
conda create -n phy

you could try clean yout conda installation
conda clean --all
also you could try to update conda 
conda update --prefix PATH_TO_CONDA anaconda
where PATH_TO_CONDA is the path where you have your base conda

rossant, 2019-06-13:
you could try installing the dev version, which should be easier to install as it uses more up-to-date dependencies

closed 2019-06-14 by rossant