cortex-lab/phy #833: Changing default number of coloured channels

Zoe0793, 2019-06-08:
Hello, 
I am using template-gui to visualise spike sorted data from an 8 recording site * 4 shank probe. The data is analysed for each shank seperately, making each cluster have a maximum of 8 channels contributing to it. However, I can always see 16 highlighted channels for each cluster, this makes it confusing to determine which spike belongs to which cluster. Can you please tell me how to change the default number of coloured channels from 16 to 8. 
Many thanks, 
Zoe

rossant, 2019-06-11:
Hi,
I'm currently working on a new version of phy. On this version, the following plugin works:

```
# to put in your .phy/phy_config.py
from phy import IPlugin

class MyPlugin(IPlugin):
    def attach_to_controller(self, controller):
        controller.model.n_closest_channels = 8
```

I think it should also work on the previous version.
Some guidance about the plugins can be found here:
https://phy.readthedocs.io/en/latest/customization/#changing-the-number-of-spikes-in-views

rossant, 2019-06-13:
please reopen if needed!

closed 2019-06-13 by rossant

Zoe0793, 2019-06-14:
Hi, 
Thanks for the reply. I can get the Hello World! e.g. plug in to work but the following error occurs with the plugin above:  An error occured... in: 'TemplateController' object has no attribute 'model'.

rossant, 2019-06-14:
is this on the `dev` branch on GitHub? You can install it with `pip install git+https://github.com/cortex-lab/phy.git@dev --upgrade`. You also need phylib: `pip install git+https://github.com/cortex-lab/phylib.git --upgrade`

Zoe0793, 2019-07-04:
Hello,
I am sorry for the delayed reply, I have been on holiday and the work
computers have being upgraded to windows 10 so I have had to start from
fresh. I have installed anaconda and created the environment (and added to
path) but now phy is not being recognised as an internal or external
command so I cannot run phy template-gui .. Do you know why this may be?
Thanks,
Zoe

rossant, 2019-07-05:
Could you activate your environment and do `python -c "import phy"` ? What is the output ?
If the import fails, try typing this:
```
pip install git+https://github.com/cortex-lab/phy.git@dev
pip install git+https://github.com/cortex-lab/phylib.git
```

Zoe0793, 2019-07-05:
Thanks for the prompt reply, the first command did not work and the output
was:

Traceback (most recent call last):
  File "<string>", line 1, in <module>
ModuleNotFoundError: No module named 'phy'

The second commands did something, but has generated new errors...
Traceback (most recent call last):
  File "c:\users\kishka\anaconda3\envs\phy2\lib\runpy.py", line 193, in
_run_module_as_main
    "__main__", mod_spec)
  File "c:\users\kishka\anaconda3\envs\phy2\lib\runpy.py", line 85, in
_run_code
    exec(code, run_globals)
  File "C:\Users\kishka\Anaconda3\envs\phy2\Scripts\phy.exe\__main__.py",
line 5, in <module>
  File
"c:\users\kishka\anaconda3\envs\phy2\lib\site-packages\phy\__init__.py",
line 21, in <module>
    from .utils.config import load_master_config
  File
"c:\users\kishka\anaconda3\envs\phy2\lib\site-packages\phy\utils\__init__.py",
line 8, in <module>
    from .plugin import IPlugin, attach_plugins
  File
"c:\users\kishka\anaconda3\envs\phy2\lib\site-packages\phy\utils\plugin.py",
line 20, in <module>
    from .config import load_master_config
  File
"c:\users\kishka\anaconda3\envs\phy2\lib\site-packages\phy\utils\config.py",
line 13, in <module>
    from traitlets.config import Config, PyFileConfigLoader,
JSONFileConfigLoader
ModuleNotFoundError: No module named 'traitlets'.

Zoe0793, 2019-07-05:
When I say the second commands 'did something'. It says that both phy and
the phylib were successfully built and installed.

rossant, 2019-07-05:
Okay, for some reason the dependencies were not successfully installed with the environment. Do:

```
pip install colorcet pyopengl requests traitlets qtconsole tqdm joblib click
```

Hopefully that will do it

Zoe0793, 2019-07-05:
Yes, that has fixed it, thanks. Unfortunately, the plugin does not work and
12 channels are being highlighted instead of the desired 8. I put the
plugin directly into phy_config as suggested.

rossant, 2019-07-05:
- Delete the `.phy` subdirectory in the data directory, as well as `phy.log`, and try again?
- If that failed, could you send me the `phy.log` or copy-paste it here?

Zoe0793, 2019-07-05:
Great! All working now, thank-you for your time, Cyrille.