cortex-lab/phy #836: Aborted (core dumped) on Ubuntu 19.04

m-beau, 2019-06-09:
OS:
Ubuntu 19.04
Python 3.7.3

Versions installed:
cortexlab/phy.git@dev
cortexlab/phylib.git

Steps to reproduce the bug:
phy template-gui params.py -> the qt gui opens
Try to resize the window or to click on something -> crashed occurs

Log:
17:33:44.893 [D] model:455            Loading spike clusters.
17:33:44.997 [D] model:478            Loading templates.
17:33:45.042 [D] model:495            Templates are dense.
17:33:45.042 [D] model:501            Loading the whitening matrix.
17:33:45.043 [D] model:505            Loading the inverse of the whitening matrix.
17:33:45.044 [D] model:99             Loading traces at `/media/maxime/Npxl_data2/DK152-153/DK152_190416day1_Probe1_run1/DK152_190416_Probe1_run1_g0_t0.imec.ap.bin`.
17:33:45.045 [D] model:527            Loading features.
17:33:50.187 [D] model:537            Features are sparse.
17:33:50.187 [D] model:555            Loading template features.
17:33:51.896 [D] model:565            Template features are sparse.
17:33:51.897 [D] model:344            Load `/media/maxime/Npxl_data2/DK152-153/DK152_190416day1_Probe1_run1/cluster_group.tsv`.
17:33:51.897 [D] _misc:187            Read /media/maxime/Npxl_data2/DK152-153/DK152_190416day1_Probe1_run1/cluster_group.tsv.
17:33:52.027 [D] context:67           Initialize joblib cache dir at `/media/maxime/Npxl_data2/DK152-153/DK152_190416day1_Probe1_run1/.phy`.
17:33:52.027 [D] context:68           Reducing the size of the cache if needed.
17:33:52.171 [D] config:34            Load config file `/home/maxime/.phy/phy_config.py`.
17:33:52.172 [D] plugin:127           Loading 0 plugins.
17:33:52.507 [D] context:136          Save data to `/media/maxime/Npxl_data2/DK152-153/DK152_190416day1_Probe1_run1/.phy/spikes_per_cluster.pkl`.
17:33:52.702 [D] gui:143              Creating GUI.
17:33:52.705 [D] state:129            Could not copy non-existing default state file /home/maxime/anaconda3/envs/phy2/lib/python3.7/site-packages/phy/apps/template/static/state.json.
17:33:55.749 [D] gui:354              Add view ClusterView to GUI.
17:33:55.750 [D] gui:354              Add view SimilarityView to GUI.
17:33:55.772 [D] gui:354              Add view WaveformView to GUI.
17:33:55.773 [D] base:187             Set state for WaveformView.
17:33:55.807 [D] trace:294            Redraw the entire trace view.
17:33:56.182 [D] gui:354              Add view TraceView to GUI.
17:33:56.183 [D] base:187             Set state for TraceView.
17:33:56.210 [D] gui:354              Add view FeatureView to GUI.
17:33:56.210 [D] base:187             Set state for FeatureView.
17:33:56.218 [D] gui:354              Add view TemplateFeatureView to GUI.
17:33:56.219 [D] base:187             Set state for TemplateFeatureView.
17:33:56.223 [D] gui:354              Add view CorrelogramView to GUI.
17:33:56.223 [D] base:187             Set state for CorrelogramView.
17:33:56.231 [D] gui:354              Add view AmplitudeView to GUI.
17:33:56.232 [D] base:187             Set state for AmplitudeView.
17:33:58.517 [D] gui:354              Add view RasterView to GUI.
17:33:58.518 [D] base:187             Set state for RasterView.
17:33:58.526 [D] gui:354              Add view TemplateView to GUI.
17:33:58.527 [D] base:187             Set state for TemplateView.
17:33:58.538 [D] gui:354              Add view ISIView to GUI.
17:33:58.538 [D] base:187             Set state for ISIView.
17:33:58.551 [D] gui:354              Add view FiringRateView to GUI.
17:33:58.551 [D] base:187             Set state for FiringRateView.
17:33:58.563 [D] gui:354              Add view AmplitudeHistogramView to GUI.
17:33:58.563 [D] base:187             Set state for AmplitudeHistogramView.
17:33:58.564 [D] gui:192              Load the geometry state.
malloc_consolidate(): invalid chunk size
Aborted (core dumped)

rossant, 2019-06-09:
Two things to try (see https://github.com/openstenoproject/plover/issues/1042)
- uninstall conda pyqt5 and try to install it from pip
- line 313 of `phy.gui.qt.py` : `QT_APP = QApplication(['-'] + sys.argv)`

m-beau, 2019-06-10:
line 313 of phy.gui.qt.py : QT_APP = QApplication(['-'] + sys.argv)

worked for this error, but lead to a new one - again, the gui opens fine until I try to resize its window -> then it freezes:

rossant, 2019-06-10:
do you have the same error message with the other proposed solution? (other install procedure for PyQt5)

rossant, 2019-06-10:
Also, it might be a graphics driver problem : try to install the latest version? What is your graphics card model?

m-beau, 2019-06-10:
So, while keeping line 313 of phy.gui.qt.py : QT_APP = QApplication(['-'] + sys.argv),

I uninstalled pyqt5 with conda and installed it with pip (along with PyQtWebEngine because PyQt5.QtWebEngineWidgets was not found, it is apparently independent from the basic pyqt5 on PyPi)

But it kept crashing randomly (I could resize a window, select a unit, change the ccg bin before the core was dumped again) so I laboriously reinstalled my driver (Nvidia GeForce GTX 1080 titanX, driver 418.56).

Now it works, but it is impossible to resize the windows. This issue can be closed.

closed 2019-06-10 by m-beau