cortex-lab/phy #837: General gui windows display issues

m-beau, 2019-06-10:
It is only possible to resize windows vertically, not horizontally.

Related topic, the limit to zoom out on most windows is too constraining: one would like to shrink the ACG or ISI histogram smaller than the frame of their respective windows, but this is not possible.

OS:
Ubuntu 19.04

Version installed:
cortexlab/phy@dev
cortexlab/phylib

Steps to reproduce the bug:
Launch phy, try to resize windows.

rossant, 2019-06-10:
> It is only possible to resize windows vertically, not horizontally.

Are you talking about the main GUI window, or individual dock widgets?

> one would like to shrink the ACG or ISI histogram smaller than the frame of their respective windows,

Sure, just curious to know why you'd need to zoom out more ? Is it because you cannot resize the windows, or because some visual elements seem to be located outside the default viewport ?

rossant, 2019-12-24:
should be fixed, feel free to reopen if not

closed 2019-12-24 by rossant