cortex-lab/phy #838: Save Channel (from ClusterView)

Bionerdess, 2019-06-13:
Hi,

I would like to save the peak channel of a cluster like the group gets saved into the cluster_group.tsv file. Preferrably I'd like to just add the Channel as column into that file. But I can't figure out how to do it. I'm sure it is super easy to do that. 

Please help! 
Thank you!!!

rossant, 2019-06-13:
good idea, done in [a plugin example](https://phy.readthedocs.io/en/latest/plugins/#saving-cluster-metadata-in-a-tsv-file), using the version in the dev branch

closed 2019-06-13 by rossant

Bionerdess, 2019-06-17:
I managed to get the plugin to attach/load when loading the template-gui, but it is still not saving the channel.
What might I be doing wrong? (I copied basically the same settings as in another plugin (Amplitude Histogram) we're using)
Thank you!

rossant, 2019-06-17:
Have you used File>Save or Ctrl+S? (the plugin is activated when saving)
If it still is not working, could you post here the contents of your `~ /.phy/phy_config.json`?

Bionerdess, 2019-06-17:
I generally use Ctrl+S. I see the cluster_group.tsv get saved and \spike_clusters.npy`. and \spikes_per_cluster.pkl`. But not the cluster_channel.tsv it should show as well.

There's only a phy_config.py; here it is

# You can also put your plugins in ~/.phy/plugins/.

from phy import IPlugin

try:
    import phycontrib
except:
    pass

# Plugin example:
#
# class MyPlugin(IPlugin):
#     def attach_to_cli(self, cli):
#         # you can create phy subcommands here with click
#         pass

c = get_config()
c.Plugins.dirs = [r'C:/Users/LabAdmin/.phy/plugins/']
c.TemplateGUI.plugins = ['PeakChannel','AmplitudeHistogram']

rossant, 2019-06-17:
Replace `@gui.connect_` by `@connect`

Bionerdess, 2019-06-18:
That's where I started from, but I get the error that 'connect' is not defined. 
But when I keep the "from phy import IPlugin, connect", it gives me that the plugin can't be found.

    @connect
NameError: name 'connect' is not defined

09:24:11 [W] __init__:36          The plugin PeakChannel couldn't be found.
09:24:11 [D] __init__:40          Attached plugin AmplitudeHistogram.

rossant, 2019-06-18:
The imports are not correct: please use the plugin exactly as it is given here https://phy.readthedocs.io/en/latest/plugins/#saving-cluster-metadata-in-a-tsv-file

Bionerdess, 2019-06-18:
That's what I tried at first, but it gives me the error that the plugin couldn't be found. 
The model.py file is in this directory: 
\Miniconda3\envs\phy\Lib\site-packages\phycontrib\template

I get the same error when I keep 
"import IPlugin, connect"
with either "phylib.io.model" or "phycontrib.template.model"

the only version with which it actually loads the plugin is 
"from phy import IPlugin
from phycontrib.template.model import save_metadata"

rossant, 2019-06-18:
I think you should install phy 2 in a fresh conda environment so that you don't get conflicts with phycontrib (which is deprecated).
Create a new conda environment named `phy2` (for example with `conda create -n phy2 numpy scipy matplotlib pyqt
`) and use the instructions here: https://github.com/cortex-lab/phy#developer-installation-instructions

Bionerdess, 2019-06-18:
Now it is complaining about the file name not being in the right format.
    s = self._ext_to_normal(_getfinalpathname(s))
OSError: [WinError 123] The filename, directory name, or volume label syntax is incorrect: 'H_*.bin'
[0m

Wouldn't it be easier instead of saving a whole new tsv, just saving the whole ClusterView into the cluster_group.tsv? (But I can't find where it saves only the group and cluster_id into the metadata to change that)

Thank you for all your help btw! I'm sorry I'm so lost and keep bugging you.

rossant, 2019-06-18:
Actually the whole cluster view should be already saved in `cluster_info.tsv` :) so that should give you what you need.
But that doesn't change the fact that we should ensure that the plugin example works correctly.

Would you mind sending me a more complete error message, along with your `params.py`?

(dw, thank you for your patience and for helping me fix bugs...)

Bionerdess, 2019-06-18:
I only get the cluster_group.tsv 
Here is a screenshot of all files in a session that has been phy-ed.
[image]

I'm starting to feel like I have a different version of phy?

Here is the complete error message after installing phy2
[image]

and params.py

dat_path = 'H_*.bin'
n_channels_dat = 64
dtype = 'int16'
offset = 0
sample_rate = 25000.
hp_filtered = False

rossant, 2019-06-18:
'H_*.bin' is not a valid file name is it?

Bionerdess, 2019-06-18:
It's supposed to load the file that starts with H_ and ends with .bin and a wildcard (*) in between.

rossant, 2019-06-18:
Ah, this syntax doesn't work here, you could perhaps do something like `dat_path = glob.glob('H_*.bin')` (add `import glob` at the top)

Bionerdess, 2019-06-21:
Ok, so I uninstalled and reinstalled Miniconda and phy. I think I had the master version of phy installed before, from 3 years ago. with the new version now I get the cluster_info file. but neither plugins are loading properly. 
the amplitude histogram is now vertical instead of horizontal and doesn't have the gaussian fit over it and doesn't tell me the % spike cut off anymore.

rossant, 2019-06-21:
Do you have an error message with the peak channel plugin?

Regarding the amplitude histogram plugin, it hasn't been updated yet to the new version. What you see is now a builtin amplitude view with a new vertical histogram. If you'd like to see more features in this view, you can open a new issue.

Bionerdess, 2019-06-21:
I just saw that the Peak channel plugin attaches now and saves the cluster_channel.tsv file.

Ok, thank you, I'll open the issue about the amplitude histogram. I also just realized that when you split a cluster, and the cluster has previously been identified as "good", both new clusters will then be classified as "good". So I'll open one about that too.

Thank you so much for all your help!