cortex-lab/phy #842: Changes to AmplitudeView

mswallac, 2019-06-17:
Hey Cyrille,

Below are some of the modifications I made to implement the changes to AmplitudeView which Mike Economo proposed (I'm the other Mike cc'd on those emails).

In reference to these two items:

`apps/template/gui.py

    def get_amplitudes(self, cluster_ids, chan=None, load_all=False):
        n = self.n_spikes_amplitudes if not load_all else None
        m = self.model
        data_bounds = [0., 0., m.duration, None]
        full_l = (self.supervisor.clustering.cluster_ids)
        if chan is None:
            chan=self.get_best_channel(cluster_ids[0])
        bunchs = []
        for cid in full_l:
            if chan in self.get_best_channels(cid):
                spike_ids = self.selector.select_spikes([cid], n)
                
                #raw amplitude (mV) -- very slow because doing max-min every time
                #data = m.get_waveforms(spike_ids,[chan])
                #y = np.subtract(np.amax(data,axis=1),np.amin(data,axis=1))
                #x = m.spike_times[spike_ids]
                
                #PC0
                y = self.get_features(cid,[chan]).data[:,0,0]
                x = self.get_spike_times(cid).data
                
                # Projection onto selected Cluster template -- breaks on new cluster assignment (merge/split)
                #y = m.get_template_features(spike_ids)[:,cluster_ids[0]]
                #x = m.spike_times[spike_ids]
                
                if not (cid in cluster_ids):
                    bunchs.append(Bunch(x=x, y=y, spike_ids=spike_ids,
                        data_bounds=data_bounds, br=True, col=[1,1,1,.15]))
                else:
                    bunchs.append(Bunch(x=x, y=y, spike_ids=spike_ids, 
                        data_bounds=data_bounds, br=False, col=cluster_ids.index(cid)))
    
        ymax = max(b.y.max() for b in bunchs)
        for bunch in bunchs:
            bunch.data_bounds[-1] = ymax
        return bunchs

clustering/views/scatter.py

    # in imports:
    import phy.apps.template.gui as pat
    #to ensure that I only respond to channel click with AmplitudeView in on_channel_click

    #MODIFIED
    def _plot_points(self, bunchs, data_bounds):
        ms = self._default_marker_size
        xmin, ymin, xmax, ymax = data_bounds
        self.visual.reset_batch()
        for i, d in enumerate(bunchs):
            x, y = d.x, d.y
            assert x.ndim == y.ndim == 1
            assert x.shape == y.shape
            if hasattr(d,'br'):
                if d.br:
                    color = d.col
                else:
                    color = selected_cluster_color(d.col,0.75)
            else:
                color = selected_cluster_color(i,0.75)

            # Create one visual per cluster.
            self.visual.add_batch_data(x=x, y=y, color=color, size=ms, data_bounds=data_bounds)
        self.canvas.update_visual(self.visual)

    #ADDED
    def on_channel_click(self, sender=None, channel_id=None, key=None, button=None):
        if channel_id is not None and isinstance(self,pat.AmplitudeView):
            bunchs = self.coords(self.cluster_ids,chan=channel_id)
            if bunchs is None:
                self.visual.hide()
                self.canvas.update()
                return
            self.visual.show()
            self.data_bounds = self._get_data_bounds(bunchs)
    
            self._plot_points(bunchs, self.data_bounds)
            self.canvas.axes.reset_data_bounds(self.data_bounds)
            self.canvas.update()
    
    #MODIFIED
    def attach(self, gui):
        super(ScatterView, self).attach(gui)
        connect(self.on_request_split)
        connect(self.on_channel_click)
`

mswallac, 2019-06-17:
Accidentally posted incomplete issue, will update.

rossant, 2019-06-17:
@mswallac Thanks. FYI you can directly edit your comment on this issue instead of creating a new one

rossant, 2019-06-17:
Excellent, thanks! I'll take it from here and integrate these features properly in phy.

Regarding the different things to show in the amplitude view: one way would be to switch between different options in the same view. But another option would be to use *different* amplitude views, each showing a different variable on the y axis. You could stack these different amplitude views at the same position in the GUI, which would automatically create tabs for each, allowing you to switch between these views. I think that would be a bit easier to implement, would that work for you?

mswallac, 2019-06-17:
Sorry for the incomplete comment edit, the current state of the comment should be much clearer. I think that the multiple window implementation would be perfect! That would have the additional benefit of being able to view those features simultaneously.

rossant, 2019-06-20:
Thank you for the details and for the prototype, it was really helpful for implementing these features properly. Your proposal is now part of phy natively and I believe it will be useful to many users!

I ended up implementing just a single amplitude view where you can toggle betweel `template`, `raw`, and `feature` amplitude types by pressing `a`. You can still split in that view. Histograms are shown as well. And, for the `feature` type, you can choose the dimension from the feature view with `alt+left click` for the x axis, and `alt+right click` for the y axis.

Please try it and let me know if you have any issues or suggestions.

[image]

closed 2019-06-20 by rossant

TomBugnon, 2021-06-24:
Hi @rossant 
I'm trying to use the 'raw' amplitude view but the available amplitude types are 'feature', 'template_feature' and 'template', any idea what's going on?