cortex-lab/phy #843: ImportError: No module named 'phylib'

Robertooooooo, 2019-06-18:
Hi, 
I get this error when I do phy template-gui params.py. 
`  File "c:\users\roberto\anaconda3\envs\phy\lib\runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "c:\users\roberto\anaconda3\envs\phy\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "C:\Users\Roberto\Anaconda3\envs\phy\Scripts\phy.exe\__main__.py", line 5, in <module>
  File "c:\users\roberto\anaconda3\envs\phy\lib\site-packages\phy\__init__.py", line 18, in <module>
    from phylib.utils import Bunch
ImportError: No module named 'phylib'`
I see phylib here https://github.com/cortex-lab/phylib but what should I do with it? 

Thanks!

rossant, 2019-06-18:
Do `pip install git+https://github.com/cortex-lab/phylib.git`

Robertooooooo, 2019-06-18:
Still get the same error :/

rossant, 2019-06-18:
Have you done : conda activate phy  before installing phylib ?

What is the output of
python -c "import phylib"

Robertooooooo, 2019-06-18:
that error is gone thanks to conda activate phy. but after that i got
`ImportError: No module named 'PyQt5'`
I solved that with 'pip install PyQt5', then I got
`ImportError: No module named 'PyQt5.QtWebEngineWidgets'`
solved it with 'pip install pyqtwebengine'. Now I´m stuck at
`  File "c:\users\roberto\anaconda3\envs\phy\lib\site-packages\matplotlib\backends\backend_qt5agg.py", line 15, in <module>
    from .backend_qt5 import QtCore
  File "c:\users\roberto\anaconda3\envs\phy\lib\site-packages\matplotlib\backends\backend_qt5.py", line 27, in <module>
    import matplotlib.backends.qt_editor.figureoptions as figureoptions
  File "c:\users\roberto\anaconda3\envs\phy\lib\site-packages\matplotlib\backends\qt_editor\figureoptions.py", line 17, in <module>
    import matplotlib.backends.qt_editor.formlayout as formlayout
  File "c:\users\roberto\anaconda3\envs\phy\lib\site-packages\matplotlib\backends\qt_editor\formlayout.py", line 58, in <module>
    from matplotlib.backends.qt_compat import QtGui, QtWidgets, QtCore
  File "c:\users\roberto\anaconda3\envs\phy\lib\site-packages\matplotlib\backends\qt_compat.py", line 124, in <module>
    from PyQt4 import QtCore, QtGui
ValueError: PyCapsule_GetPointer called with incorrect name`

rossant, 2019-06-18:
Actually this is not the right way to install the latest version. Please create a new conda environment named `phy2` (for example with `conda create -n phy2 numpy scipy matplotlib pyqt
`) and use the instructions here: https://github.com/cortex-lab/phy#developer-installation-instructions

Robertooooooo, 2019-06-18:
ok, I created a new environment phy2 but now after doing `conda activate phy2` if i try to clone the repository I still get
 `fatal: destination path 'phy' already exists and is not an empty directory.` 

I´ll try to reinstall anaconda tomorrow to get the master phy working again.

rossant, 2019-06-18:
I think this is because there is already a phy subdirectory containing the previous version
rename it to phy1 and try again

Robertooooooo, 2019-06-18:
Thanks for the quick answers :)

Robertooooooo, 2019-06-19:
Renaming did the trick!
Thanks !

closed 2019-06-19 by Robertooooooo