cortex-lab/phy #845: Phy2 list of issues

dimokaramanlis, 2019-06-21:
Hi phy2 team,

thank you for the excellent software. We just installed the dev version of phy2, and we noticed the following:

1) Some of the features described in the documentation, such as RasterView, AmplitudeHistogramView or TemplateView are missing.

2) The "Go To" function of the TraceView doesn't work. Gives error "Invalid function arguments: expecting 1 but got 0[0m".

3) In the AmplitudeView, what is the scale of the "raw" amplitudes? We read the docs, but it doesn't seem to match the int16 raw values. We get values between 0-20 usually.

4) In ClusterView, after moving to a next cluster with the keyboard (down key), the panel doesn't scroll down. After going beyond the clusters visible in the panel, then one can't see which cluster is selected (unless one scrolls with the mouse).

5) It would be great if one could select the starting time in the FiringRate view as well. Instead of only having the option to look at [0, 20000] or [0 1000], it would be useful to look at [19000 20000].

6) In WaveformView, sometimes the templates and spike traces don't have the same amplitude. So, when toggling between them (with key W), the spike traces look smaller than the templates. The same happens when toggling between spikes and mean waveforms (key M).

We look forward to the full release!

rossant, 2019-06-21:
Thanks for your feedback!

> 1. Some of the features described in the documentation, such as RasterView, AmplitudeHistogramView or TemplateView are missing.

They're no longer added by default since they may be a bit heavy on large datasets, but you can add them with `View > New view`.

> 2\. The "Go To" function of the TraceView doesn't work. Gives error "Invalid function arguments: expecting 1 but got 0�[0m".

Thanks! Fixed now.

> 3\. In the AmplitudeView, what is the scale of the "raw" amplitudes? We read the docs, but it doesn't seem to match the int16 raw values. We get values between 0-20 usually.

Raw amplitudes are taken from the waveforms, which themselves come from the raw data. I'm not sure why you see this discrepancy, could you send me a sample of your dataset perhaps?

> 4\. In ClusterView, after moving to a next cluster with the keyboard (down key), the panel doesn't scroll down. After going beyond the clusters visible in the panel, then one can't see which cluster is selected (unless one scrolls with the mouse).

This is a know issue, I'm not sure how to fix this with the way Qt handles HTML views. This feature was implemented before but it didn't work well for some reason. I'll keep looking... (feel free to open a new issue specifically about this)

> 5\. It would be great if one could select the starting time in the FiringRate view as well. Instead of only having the option to look at [0, 20000] or [0 1000], it would be useful to look at [19000 20000].

Good idea, I'll do it!

> 6\. In WaveformView, sometimes the templates and spike traces don't have the same amplitude. So, when toggling between them (with key W), the spike traces look smaller than the templates. The same happens when toggling between spikes and mean waveforms (key M).

Might be related to (3)

rossant, 2019-07-05:
> This is a know issue, I'm not sure how to fix this with the way Qt handles HTML views. This feature was implemented before but it didn't work well for some reason. I'll keep looking... (feel free to open a new issue specifically about this)

Done!

All should have been done, except perhaps https://github.com/cortex-lab/phy/issues/850 
Closing this issue now

closed 2019-07-05 by rossant