cortex-lab/phy #846: how to reset waveform view

bfpcruz, 2019-06-26:
hey everyone,
I somehow ended up with the waveform visualizer bugged (probably something to do with scale or limits) and i cant seem to get it back to a sensible range. Any way to reset the settings?

[image]

rossant, 2019-06-26:
I'll add an option soon. In the meantime, delete these 2 files : `~/.phy/TemplateGUI/state.json`, and `.phy/state.json`

malinroe, 2019-07-02:
Hi, I'm also having this problem (win10). It works for a second if I delete the state.json files, but gets bugged the minute I touch the waveform visualizer.

rossant, 2019-07-02:
The scaling bug in the waveform view should be fixed in the latest version on dev.
There is no longer a scaling parameter in the feature view at the moment.

closed 2019-07-02 by rossant