cortex-lab/phy #847: ACG doesn't show isi violation within 0.5 ms?

susuchen66, 2019-06-29:
Hi Cyrille,

I'm not sure if this is a bug or actually intended, in both the old version and new version of Phy, the autocorrelogram doesn't show any ISI violation within ~0.5 ms bin. Now with the ISI distribution added to Phy2, you can see in the example attached, there's a significant amount of ISI violation in this unit (as a result of double counted spikes) but all gets removed in ACG. Could you please look into the display? It might be misleading given people taking into account ACG a lot during manual curation. 
A correct ACG plot is also enclosed.
Many thanks!
[image]
[image]

rossant, 2019-06-29:
At the moment the central peak is removed to avoid the artifacts due to the algorithm but that could be reconsidered, in the meantime does changing the binning to e.g. 0.1 ms fixes the problem for you?

susuchen66, 2019-06-29:
Thanks Cyrille, after changing the binning to 0.1 ms or 0.2 ms, the central peak would show up.

closed 2019-07-02 by rossant

susuchen66, 2019-07-06:
Hi Cyrille, could you fix this, since it's misleading to not show the central peak, regardless of which bin size to use. 
Thanks!

rossant, 2019-07-10:
Yes you're right, this is now fixed on the latest version of phylib. Thanks for your feedback!