cortex-lab/phy #848: Individual views hang

marius10p, 2019-06-30:
This happens relatively often, to one view at a time, and can happen to either one of the views while the other ones are fine. It outputs this message when one view crashes:

QBasicTimer::start: Timers cannot be started from another thread

I am on Win10 and I had to "pip install PyQtWebEngine" to make it work. Closing and re-opening the crashed view works fine.

rossant, 2019-07-01:
Does it still happen at the same frequency if you use the following plugin (to put in `~/.phy/phy_config.py`)

```python
from phy import IPlugin, connect

c = get_config()

class MyPlugin(IPlugin):
    def attach_to_controller(self, controller):
        @connect
        def on_gui_ready(sender, gui):
            gui._enable_threading = False

c.TemplateGUI.plugins = ['MyPlugin']
```

This disables multithreading for view updates.

marius10p, 2019-07-02:
It no longer breaks, but it's noticeably slower, about as slow as Phy 1.0.

rossant, 2019-07-02:
Okay I'll try to investigate further. Note that you can disable auto-update of certain views, menu Views>SomeView>Toggle auto update

rossant, 2019-08-09:
hopefully fixed with the latest version by 6b08b08e52f473541d9149024a2fcfc5fa699f27

closed 2019-08-09 by rossant