cortex-lab/phy #849: hidden waveform views?

haiderlab, 2019-07-02:
hello phy team,

since the phy update from dev version, i often can't draw into or raise the default WaveformView.  Adding new WaveformViews (2), (3), etc. sometimes alleviates this, but keyboard shortcuts (overlap, toggle mean) then don't work and 'assertion errors' become common when running the wizard from these additional WaveformViews while interacting in other views (Raster, Feature).  This is only solved by killing all views except the ClusterView, and restarting. 

is there a simple way to reset to defaults / kill waveform views, so that the default view is drawn into / visible?

thanks

rossant, 2019-07-03:
Is this still happening on the very latest version on the dev branch?

haiderlab, 2019-07-03:
Yes, it is happening on the latest version

rossant, 2019-07-03:
Is this a duplicate of https://github.com/cortex-lab/phy/issues/848 ?
What is your OS?

haiderlab, 2019-07-03:
Windows 10.  Different from #848, and I haven't yet seen the same error described there.  I tried pip install PyQtWebEngine and this hasn't changed the behavior.  The same has just happened to the original FeatureView, I can no longer see it , select it, or maximize it...

haiderlab, 2019-07-03:
Posted a few screencaps here if helpful:
https://www.dropbox.com/sh/9ye71hzelopmisz/AAAWZZy0MhdU0R1kMdAzV9dja?dl=0

Image descriptions:
(1) Default view upon launching phy.  Menu shows WaveformView, and additional Views (2) and (3).  None of these are visible.  Also note that a cluster is selected and drawn into Feature View.  Launching wizard shows blue and next best red cluster in Feature view, but no WaveformViews.
(2) Addition of new WaveformView (3), visible at bottom.
(3) Docked WaveFormView(3) now shows clusters, but keyboard shortcuts are not active.  Toggling from dropdown menu works for a while, but eventually throw an AssertionError as described above, and windows hang / crash.
(4) Screencap from just two days ago with everything in working order (all default views, no additional views).  One thing that is different is that the file path is visible in the TemplateGUI menu bar at top left.

rossant, 2019-07-03:
Okay thanks, could you delete `~/.phy/TemplateGUI/state.json` to reset the window layout and try again?

haiderlab, 2019-07-03:
this appears to have solved the issue, thank you!

rossant, 2019-07-05:
Added a `--clear-config` option to the `phy` command to delete the global and local `state.json` and reset the GUI layout on startup.

closed 2019-07-05 by rossant

ijunhyuk, 2024-03-22:
As of 2024 (phy version 2.0b5), I have resolved this issue with the following command. This command initialized the arrangement of View panels.

phy template-gui  params.py --clear-state