cortex-lab/phy #850: Scaling mismatch between raw data and mean and template

yger, 2019-07-03:
This is not crucial, but there is a scaling mismatch in the WaveformView between the raw data and the mean/template. Since it may only be happening for data exported with spyking-circus, maybe something related to the whitening. Is the same behaviour observed in the test data we gave you?

rossant, 2019-07-03:
Could you post a screenshot with the test data I have, showing which clusters are selected so that I double check?

yger, 2019-07-08:
Here is one example, but I can give you more. Sometimes, this looks more striking, but I just realized that maybe you are rescaling somehow the data while switching between these plots

[image]

[image]

rossant, 2019-08-11:
Should be fixed by @mswallac's pull request

closed 2019-08-11 by rossant