cortex-lab/phy #851: Fail to relod data

yagui, 2019-07-10:
When I run phy for the first time over a new sorted data it runs ok.
After I close it, if I want to load again this data, I have to delete .phy folder
Otherwise I get this error
```
Traceback (most recent call last):
  File "/home/ariel/virtualenvs/spyking/.spyking/lib/python3.6/site-packages/phy/gui/widgets.py", line 249, in emitJS
    self._debouncer.submit(emit, *args)
  File "/home/ariel/virtualenvs/spyking/.spyking/lib/python3.6/site-packages/phy/gui/qt.py", line 508, in submit
    self.trigger()
  File "/home/ariel/virtualenvs/spyking/.spyking/lib/python3.6/site-packages/phy/gui/qt.py", line 523, in trigger
    f(*args, **kwargs)
  File "/home/ariel/virtualenvs/spyking/.spyking/lib/python3.6/site-packages/phylib/utils/event.py", line 140, in emit
    res.append(f(sender, *args, **kwargs))
  File "/home/ariel/virtualenvs/spyking/.spyking/lib/python3.6/site-packages/phy/cluster/supervisor.py", line 836, in _clusters_selected
    emit('select', self, self.selected, **kwargs)
  File "/home/ariel/virtualenvs/spyking/.spyking/lib/python3.6/site-packages/phylib/utils/event.py", line 140, in emit
    res.append(f(sender, *args, **kwargs))
  File "/home/ariel/virtualenvs/spyking/.spyking/lib/python3.6/site-packages/phy/apps/base.py", line 904, in on_select
    view.update_color(selected_clusters=cluster_ids)
  File "/home/ariel/virtualenvs/spyking/.spyking/lib/python3.6/site-packages/phy/cluster/views/template.py", line 179, in update_color
    return self.plot()
  File "/home/ariel/virtualenvs/spyking/.spyking/lib/python3.6/site-packages/phy/cluster/views/template.py", line 198, in plot
    bunchs = self.get_clusters_data()
  File "/home/ariel/virtualenvs/spyking/.spyking/lib/python3.6/site-packages/phy/cluster/views/template.py", line 140, in get_clusters_data
    bunchs = self.templates(self.all_cluster_ids)
  File "/home/ariel/virtualenvs/spyking/.spyking/lib/python3.6/site-packages/phy/apps/base.py", line 413, in _get_all_templates
    for cluster_id in cluster_ids}
TypeError: 'NoneType' object is not iterable
[lagertha:25996] *** Process received signal ***
[lagertha:25996] Signal: Aborted (6)
[lagertha:25996] Signal code:  (-6)
[lagertha:25996] [ 0] /lib/x86_64-linux-gnu/libc.so.6(+0x3ef20)[0x7ffa8bad0f20]
[lagertha:25996] [ 1] /lib/x86_64-linux-gnu/libc.so.6(gsignal+0xc7)[0x7ffa8bad0e97]
[lagertha:25996] [ 2] /lib/x86_64-linux-gnu/libc.so.6(abort+0x141)[0x7ffa8bad2801]
[lagertha:25996] [ 3] /home/ariel/virtualenvs/spyking/.spyking/lib/python3.6/site-packages/PyQt5/Qt/lib/libQt5Core.so.5(_Z13qErrnoWarningPKcz+0x0)[0x7ffa324d9e2c]
[lagertha:25996] [ 4] /home/ariel/virtualenvs/spyking/.spyking/lib/python3.6/site-packages/PyQt5/QtCore.so(+0x1e1bde)[0x7ffa32dc8bde]
[lagertha:25996] [ 5] /home/ariel/virtualenvs/spyking/.spyking/lib/python3.6/site-packages/PyQt5/QtCore.so(+0x1ea6d8)[0x7ffa32dd16d8]
[lagertha:25996] [ 6] /home/ariel/virtualenvs/spyking/.spyking/lib/python3.6/site-packages/PyQt5/QtCore.so(+0x1ea802)[0x7ffa32dd1802]
[lagertha:25996] [ 7] /home/ariel/virtualenvs/spyking/.spyking/lib/python3.6/site-packages/PyQt5/Qt/lib/libQt5Core.so.5(_ZNK11QMetaMethod6invokeEP7QObjectN2Qt14ConnectionTypeE22QGenericReturnArgument16QGenericArgumentS5_S5_S5_S5_S5_S5_S5_S5_S5_+0x8aa)[0x7ffa326d356a]
[lagertha:25996] [ 8] /home/ariel/virtualenvs/spyking/.spyking/lib/python3.6/site-packages/PyQt5/Qt/lib/libQt5WebChannel.so.5(_ZN20QMetaObjectPublisher12invokeMethodEP7QObjectiRK10QJsonArray+0x700)[0x7ffa181bd370]
[lagertha:25996] [ 9] /home/ariel/virtualenvs/spyking/.spyking/lib/python3.6/site-packages/PyQt5/Qt/lib/libQt5WebChannel.so.5(_ZN20QMetaObjectPublisher13handleMessageERK11QJsonObjectP28QWebChannelAbstractTransport+0x59f)[0x7ffa181c2d5f]
[lagertha:25996] [10] /home/ariel/virtualenvs/spyking/.spyking/lib/python3.6/site-packages/PyQt5/Qt/lib/libQt5Core.so.5(_ZN11QMetaObject8activateEP7QObjectiiPPv+0x986)[0x7ffa326f2f26]
[lagertha:25996] [11] /home/ariel/virtualenvs/spyking/.spyking/lib/python3.6/site-packages/PyQt5/Qt/lib/libQt5WebChannel.so.5(_ZN28QWebChannelAbstractTransport15messageReceivedERK11QJsonObjectPS_+0x34)[0x7ffa181c8924]
[lagertha:25996] [12] /home/ariel/virtualenvs/spyking/.spyking/lib/python3.6/site-packages/PyQt5/Qt/lib/libQt5WebEngineCore.so.5(+0x87eb6a)[0x7ffa1955cb6a]
[lagertha:25996] [13] /home/ariel/virtualenvs/spyking/.spyking/lib/python3.6/site-packages/PyQt5/Qt/lib/libQt5WebEngineCore.so.5(+0x886c73)[0x7ffa19564c73]
[lagertha:25996] [14] /home/ariel/virtualenvs/spyking/.spyking/lib/python3.6/site-packages/PyQt5/Qt/lib/libQt5WebEngineCore.so.5(+0x2570d39)[0x7ffa1b24ed39]
[lagertha:25996] [15] /home/ariel/virtualenvs/spyking/.spyking/lib/python3.6/site-packages/PyQt5/Qt/lib/libQt5WebEngineCore.so.5(+0x2a247c9)[0x7ffa1b7027c9]
[lagertha:25996] [16] /home/ariel/virtualenvs/spyking/.spyking/lib/python3.6/site-packages/PyQt5/Qt/lib/libQt5WebEngineCore.so.5(+0x2a249e3)[0x7ffa1b7029e3]
[lagertha:25996] [17] /home/ariel/virtualenvs/spyking/.spyking/lib/python3.6/site-packages/PyQt5/Qt/lib/libQt5WebEngineCore.so.5(+0x2259767)[0x7ffa1af37767]
[lagertha:25996] [18] /home/ariel/virtualenvs/spyking/.spyking/lib/python3.6/site-packages/PyQt5/Qt/lib/libQt5WebEngineCore.so.5(+0x227cea7)[0x7ffa1af5aea7]
[lagertha:25996] [19] /home/ariel/virtualenvs/spyking/.spyking/lib/python3.6/site-packages/PyQt5/Qt/lib/libQt5WebEngineCore.so.5(+0x227d50d)[0x7ffa1af5b50d]
[lagertha:25996] [20] /home/ariel/virtualenvs/spyking/.spyking/lib/python3.6/site-packages/PyQt5/Qt/lib/libQt5WebEngineCore.so.5(+0x227d7da)[0x7ffa1af5b7da]
[lagertha:25996] [21] /home/ariel/virtualenvs/spyking/.spyking/lib/python3.6/site-packages/PyQt5/Qt/lib/libQt5WebEngineCore.so.5(+0x7e2a48)[0x7ffa194c0a48]
[lagertha:25996] [22] /home/ariel/virtualenvs/spyking/.spyking/lib/python3.6/site-packages/PyQt5/Qt/lib/libQt5Core.so.5(_ZN7QObject5eventEP6QEvent+0x7b)[0x7ffa326f3b0b]
[lagertha:25996] [23] /home/ariel/virtualenvs/spyking/.spyking/lib/python3.6/site-packages/PyQt5/Qt/lib/libQt5Widgets.so.5(_ZN19QApplicationPrivate13notify_helperEP7QObjectP6QEvent+0x9c)[0x7ffa2c9896fc]
[lagertha:25996] [24] /home/ariel/virtualenvs/spyking/.spyking/lib/python3.6/site-packages/PyQt5/Qt/lib/libQt5Widgets.so.5(_ZN12QApplication6notifyEP7QObjectP6QEvent+0x2b0)[0x7ffa2c9907f0]
[lagertha:25996] [25] /home/ariel/virtualenvs/spyking/.spyking/lib/python3.6/site-packages/PyQt5/QtWidgets.so(+0x36cf5e)[0x7ffa2d3e3f5e]
[lagertha:25996] [26] /home/ariel/virtualenvs/spyking/.spyking/lib/python3.6/site-packages/PyQt5/Qt/lib/libQt5Core.so.5(_ZN16QCoreApplication15notifyInternal2EP7QObjectP6QEvent+0x108)[0x7ffa326c71d8]
[lagertha:25996] [27] /home/ariel/virtualenvs/spyking/.spyking/lib/python3.6/site-packages/PyQt5/Qt/lib/libQt5Core.so.5(_ZN23QCoreApplicationPrivate16sendPostedEventsEP7QObjectiP11QThreadData+0x267)[0x7ffa326c9c77]
[lagertha:25996] [28] /home/ariel/virtualenvs/spyking/.spyking/lib/python3.6/site-packages/PyQt5/Qt/lib/libQt5Core.so.5(+0x2caf13)[0x7ffa3271ef13]
[lagertha:25996] [29] /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_dispatch+0x2e7)[0x7ffa2fd53417]
[lagertha:25996] *** End of error message ***
Aborted (core dumped)
```
Although I delete the .phy folder I don't loose my updated cluster. What is the porpoise of .phy folder? it is safe to delete everytime?

It just started happen to me after I updated to the last version. I'll try to reinstall previous version to see if I can find any difference.

rossant, 2019-07-10:
The `.phy` subfolder is a cache that makes the GUI faster during a session, and when reopening the session. It is totally safe to delete it, you just lose a bit of performance the next time you reopen the GUI. Your data is saved in the data directory, in `spike_clusters.npy` and in the tsv files.

Regarding the bug you have, I pushed a quick fix but I'm not sure that will be enough. Could you do `pip install git+https://github.com/cortex-lab/phy@dev --upgrade` and try again?

yagui, 2019-07-10:
It keeps happening
I use spyking-circus and it has a command to invoke phy. In this case the GUI shuts down.
If I try to open phy without spyking-circus command, it doesn't shut down, only it shows an error message window and continues, so it's not a big deal.

I tested also with this data
https://github.com/kwikteam/phy-data

With the last fix the error now has changed to a different line

```
13:23:38.351 [E] __init__:58          An error has occurred (TypeError): object of type 'NoneType' has no len()
Traceback (most recent call last):
  File "/home/ariel/virtualenvs/spyking/.spyking/lib/python3.6/site-packages/phy/gui/widgets.py", line 249, in emitJS
    self._debouncer.submit(emit, *args)
  File "/home/ariel/virtualenvs/spyking/.spyking/lib/python3.6/site-packages/phy/gui/qt.py", line 508, in submit
    self.trigger()
  File "/home/ariel/virtualenvs/spyking/.spyking/lib/python3.6/site-packages/phy/gui/qt.py", line 523, in trigger
    f(*args, **kwargs)
  File "/home/ariel/virtualenvs/spyking/.spyking/lib/python3.6/site-packages/phylib/utils/event.py", line 140, in emit
    res.append(f(sender, *args, **kwargs))
  File "/home/ariel/virtualenvs/spyking/.spyking/lib/python3.6/site-packages/phy/cluster/supervisor.py", line 836, in _clusters_selected
    emit('select', self, self.selected, **kwargs)
  File "/home/ariel/virtualenvs/spyking/.spyking/lib/python3.6/site-packages/phylib/utils/event.py", line 140, in emit
    res.append(f(sender, *args, **kwargs))
  File "/home/ariel/virtualenvs/spyking/.spyking/lib/python3.6/site-packages/phy/apps/base.py", line 906, in on_select
    view.update_color(selected_clusters=cluster_ids)
  File "/home/ariel/virtualenvs/spyking/.spyking/lib/python3.6/site-packages/phy/cluster/views/template.py", line 179, in update_color
    return self.plot()
  File "/home/ariel/virtualenvs/spyking/.spyking/lib/python3.6/site-packages/phy/cluster/views/template.py", line 198, in plot
    bunchs = self.get_clusters_data()
  File "/home/ariel/virtualenvs/spyking/.spyking/lib/python3.6/site-packages/phy/cluster/views/template.py", line 142, in get_clusters_data
    for cluster_rel, cluster_idx, cluster_id in self._iter_clusters():
  File "/home/ariel/virtualenvs/spyking/.spyking/lib/python3.6/site-packages/phy/cluster/views/base.py", line 364, in _iter_clusters
    for i in range(len(self.all_cluster_ids)):
TypeError: object of type 'NoneType' has no len()
```

rossant, 2019-07-10:
I pushed a new fix, could you update and try again?

yagui, 2019-07-10:
No error, It seems fixed. Thanks!

closed 2019-07-10 by yagui