cortex-lab/phy #854: Option to display all channels in waveform view

angelonc, 2019-07-19:
I've been using 32ch tetrode arrays recently, and the waveform view is a bit annoying in phy, as it will display parts of nearby tetrodes, depending on where the most active channel is.

Is there a config option I can change to increase the number of nearest channels to display (I think a previous version of phy simply showed all channels)?

rossant, 2019-07-23:
A "show all channels" option would be a good idea. In the meantime, change the `n_closest_channels` parameter in https://phy.readthedocs.io/en/latest/plugins/#changing-the-number-of-spikes-in-views

angelonc, 2019-07-24:
Thanks, making a plugin to change the default parameters did the trick!

closed 2019-07-24 by angelonc

dervinism, 2023-03-29:
It increases the number of channels but waveforms are flat on those extra channels after kilosort3.