cortex-lab/phy #856: error starting phy

siaahmadi, 2019-07-19:
Running `phy template-gui params.py` following installation instructions on Windows 10 and Python 3.7.3 gives the following error:

[image]

Please help.

rossant, 2019-07-23:
could you try to install the latest version of your graphics driver

siaahmadi, 2019-07-23:
> could you try to install the latest version of your graphics driver

I have already done that.

rossant, 2019-07-26:
What is your graphics card? Do you happen to have both an integrated Intel and a dedicated NVIDIA graphics chipset?

siaahmadi, 2019-07-26:
I only have the RTX 2070.

[image]

rossant, 2019-08-06:
could you tell me the OpenGL version displayed by the following program: http://realtech-vr.com/admin/glview ?

megkirch, 2019-08-21:
I am having the same issue. I downloaded the OpenGL Extensions Viewer and I'm using OpenGL version 3.1 with an Intel HD Graphics 3000 renderer. Is there any update on how to resolve this issue?

rossant, 2019-08-23:
I'm out of suggestions, being unable to reproduce this error on my Windows 10 machine, and not finding much help about this error message online. 
Perhaps try to install `pip install PyOpenGL-accelerate`  but I doubt it would fix the issue.
@rougier have you ever seen this bug with glumpy (my library uses `glumpy.gloo` which is based on PyOpenGL)?

rougier, 2019-08-23:
No real idea either. Maybe [this thread on stackoverflow](https://stackoverflow.com/questions/40123969/pyopengl-glclearcolor-error-in-windows-ok-in-linux) can help.

siaahmadi, 2019-08-23:
> could you tell me the OpenGL version displayed by the following program: http://realtech-vr.com/admin/glview ?

My OpenGL Driver version is 26.21.14.3602 (16-Aug-19).

megkirch, 2019-09-03:
Figured it out - it was as simple as I was plugged into the wrong port on my PC! As soon I switched to the NVidia GPU port everything worked fine. Perhaps others running into this issue may be having the same problem?

closed 2019-12-24 by rossant