cortex-lab/phy #857: polygon location mismatch from cursor position

HiroyukiKato7, 2019-07-20:
When splitting a cluster in the Feature View, there is a location mismatch between where I click and where the polygon is created.  This mismatch is negligible around the center of each plot, but it gets bigger as you click further from the center and that makes it hard to draw polygons in the desired position.

This mismatch exists only in Feature View, and not in Amplitude View or Template Feature View.  I would appreciate it if somebody could fix this.

rossant, 2019-07-23:
thanks for the bug report, I'll look into it

yagui, 2019-07-24:
What I found is that you need to click in the middle of the different plots to get a polygon point in the edge. I don't know if this helps to find the bug.
[image]

rossant, 2019-08-09:
should be fixed in the dev branch

closed 2019-08-09 by rossant