cortex-lab/phy #858: phy startup error on Windows 10 with Anaconda 3

jlee536, 2019-07-22:
Hi all,

I installed phy2 following the official installation documentation with Anaconda 3 in Windows 10. When I try to launch phy by invoking "phy template-gui params.py" after activating phy2 environment, I always get a few error messages as shown in the attached screenshots, seemingly because a few modules are missing, mainly "phy.utils.color", "sklearn", and "phycontrib". I actually installed scikit-learn using conda.
What did I miss? How can I remedy those issues? Any help will be greatly appreciated!!

[image]

[image]

[image]

angelonc, 2019-07-23:
Once you're in the phy environment, try using pip to install the missing modules. (although, it is a bit weird that phycontrib is not found)

rossant, 2019-07-23:
This is due to an incompatibility with the existing plugins, try to disable them in phy_config

rossant, 2019-08-08:
closing, please reopen if this hasn't fixed it

closed 2019-08-08 by rossant