cortex-lab/phy #859: Error starting template-gui

jaredluckas, 2019-07-22:
Hi,

I'm having issues starting up phy, but only on certain computers. I'm currently running miniconda3 on Windows 10. Each time I try to use template-gui, I run into an error that says 

> TypeError: sequence item 0: expected str instance, VertexShader found

Note that I do not run into this issue on a different computer where I have set up phy identically (though I cannot use that computer to process the data due to low RAM). As far as I know, there should be no difference between the two computers set-up wise, yet I am still getting this error. I've tried reinstalling everything only to continually get this error. Does anyone know what it might mean?

Thanks

rossant, 2019-07-23:
If you run phy with the `--debug` option on the command line, do you get a more complete error message/traceback?

JannaAarse, 2019-07-23:
Hi, 

I have the same problem, I was running it with --debug and this is the error message I get : 

(phy) C:\Temp_sorting\VR32\20190722-kilosort\New folder>phy template-gui params.py --debug
15:38:45 [D] config:46            Load config file `C:\Users\jma18\.phy\phy_config.py`.
15:38:46 [D] plugin:35            Register plugin `phycontrib.kwik_gui.gui.KwikGUIPlugin`.
15:38:46 [D] plugin:35            Register plugin `phycontrib.template.gui.TemplateGUIPlugin`.
15:38:46 [D] plugin:35            Register plugin `phycontrib.utils.precache.PrecachePlugin`.
15:38:46 [D] plugin:35            Register plugin `phycontrib.utils.saveprompt.SavePromptPlugin`.
15:38:46 [D] cli:107              Attach plugin `phycontrib.kwik_gui.gui.KwikGUIPlugin` to CLI.
15:38:46 [D] cli:107              Attach plugin `phycontrib.template.gui.TemplateGUIPlugin` to CLI.
15:38:46 [D] gui:253              Loading traces at `20190722_JAVR32_Light_session_13.bin`.
15:38:46 [D] gui:268              Loading amplitudes.
15:38:46 [D] gui:277              Loading 31098 spike clusters.
15:38:46 [D] gui:283              Loading spike templates.
15:38:46 [D] gui:289              Loading spike samples.
15:38:46 [D] gui:293              Loading templates.
15:38:46 [D] gui:299              Loading the whitening matrix.
15:38:46 [D] gui:303              Loading unwhitened templates.
15:38:46 [D] gui:321              Loading similar templates.
15:38:46 [D] gui:326              Loading channel mapping.
15:38:46 [D] gui:334              Loading channel positions.
15:38:46 [D] gui:339              Loading features.
15:38:46 [D] gui:370              Loading template features.
15:38:46 [D] gui:424              HP filtering the data for waveforms
15:38:46 [D] gui:453              Loading the cluster groups.
15:38:46 [D] context:56           Initialize joblib cache dir at `C:\Temp_sorting\VR32\20190722-kilosort\New folder\.phy`.
15:38:46 [D] context:80           Load memcache for `phy.cluster.manual.controller.get_mean_masks`.
15:38:46 [D] context:80           Load memcache for `phy.cluster.manual.controller.get_mean_features`.
15:38:46 [D] context:80           Load memcache for `phy.cluster.manual.controller.get_mean_waveforms`.
15:38:46 [D] context:80           Load memcache for `phycontrib.template.gui.get_waveforms_amplitude`.
15:38:46 [D] context:80           Load memcache for `phycontrib.template.gui.get_waveform_lims`.
15:38:46 [D] context:80           Load memcache for `phy.cluster.manual.controller.get_feature_lim`.
15:38:46 [D] context:80           Load memcache for `phy.cluster.manual.controller.get_trace_lim`.
15:38:46 [D] context:80           Load memcache for `phy.cluster.manual.controller.get_close_clusters`.
15:38:46 [D] context:80           Load memcache for `phy.cluster.manual.controller.get_probe_depth`.
15:38:46 [D] context:80           Load memcache for `phy.cluster.manual.controller.get_correlograms`.
15:38:46 [D] context:80           Load memcache for `phy.cluster.manual.controller.spikes_per_cluster`.
15:38:46 [D] context:80           Load memcache for `phycontrib.template.gui.get_amplitudes`.
15:38:46 [D] context:80           Load memcache for `phycontrib.template.gui.get_cluster_templates`.
15:38:46 [D] context:80           Load memcache for `phycontrib.template.gui.get_cluster_pair_features`.
15:38:46 [D] context:80           Load memcache for `phycontrib.template.gui._cluster_template_similarities`.
15:38:46 [D] context:80           Load memcache for `phycontrib.template.gui._sim_ij`.
15:38:46 [D] context:128          The file `C:\Temp_sorting\VR32\20190722-kilosort\New folder\.phy\new_cluster_id.json` doesn't exist.
15:38:46 [D] config:46            Load config file `C:\Users\jma18\.phy\phy_config.py`.
15:38:46 [D] gui:428              Load the GUI state from `C:\Users\jma18\.phy\TemplateGUI\state.json`.
15:38:46 [D] utils:51             Get box size for 25 points.
15:38:46 [E] cli:39               An error has occurred (TypeError): sequence item 0: expected str instance, VertexShader found
15:38:46 [D] cli:42               Traceback (most recent call last):
  File "c:\users\jma18\.conda\envs\phy\lib\runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "c:\users\jma18\.conda\envs\phy\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "C:\Users\jma18\.conda\envs\phy\Scripts\phy.exe\__main__.py", line 9, in <module>
    sys.exit(phy())
  File "c:\users\jma18\.conda\envs\phy\lib\site-packages\click\core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "c:\users\jma18\.conda\envs\phy\lib\site-packages\click\core.py", line 717, in main
    rv = self.invoke(ctx)
  File "c:\users\jma18\.conda\envs\phy\lib\site-packages\click\core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\users\jma18\.conda\envs\phy\lib\site-packages\click\core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\users\jma18\.conda\envs\phy\lib\site-packages\click\core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "c:\users\jma18\.conda\envs\phy\lib\site-packages\click\decorators.py", line 17, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "c:\users\jma18\.conda\envs\phy\lib\site-packages\phycontrib\template\gui.py", line 814, in cluster_manual
    _run_cmd('_run(params)', ctx, globals(), locals())
[... 2 lines cut ...]
  File "<string>", line 1, in <module>
[... cut ...]
    self.build()
  File "c:\users\jma18\.conda\envs\phy\lib\site-packages\phy\plot\plot.py", line 193, in build
    self.add_visual(visual)
  File "c:\users\jma18\.conda\envs\phy\lib\site-packages\phy\plot\base.py", line 283, in add_visual
    visual.on_resize(self.size)
  File "c:\users\jma18\.conda\envs\phy\lib\site-packages\phy\plot\base.py", line 81, in on_resize
    s = '\n'.join(self.program.shaders)
TypeError: sequence item 0: expected str instance, VertexShader found

Thanks for the help, 
Janna

rossant, 2019-07-23:
This is with phy1, right? Could you try with phy2 in a new conda environment?

jaredluckas, 2019-07-23:
Yes, this issue was with phy1. I was not able to resolve the problem, so I installed phy2 and it seems to run normally, with no errors.

closed 2019-07-23 by rossant

JannaAarse, 2019-07-23:
How do I install phy2 ? 
I was running --upgrade but I get this 
Requirement already up-to-date: phy in c:\users\jma18\.conda\envs\phy\lib\site-packages (1.0.9)
Requirement already up-to-date: phycontrib in c:\users\jma18\.conda\envs\phy\lib\site-packages (1.0.15)

rossant, 2019-07-23:
you'll find instructions here: https://phy.readthedocs.io/en/latest/installation/

JannaAarse, 2019-07-24:
Thanks for the instructions, I tried it and got this error message from that moment on 

(phy2) C:\Temp_sorting\VR32\20190722-kilosort\New folder>phy template-gui params.py
'phy' is not recognized as an internal or external command,
operable program or batch file.

I deleted everything I had from Phy and Anaconda, reinstalled everything and still get this error message.

JannaAarse, 2019-07-24:
I also tried everything from message #855 , running pip install for everything that I could find under pip in the environment file, but that just lead to another error message 
(phy2) C:\Temp_sorting\VR32\20190723\kilosort data>phy template-gui params.py
Usage: phy [OPTIONS] COMMAND [ARGS]...

Error: No such command "template-gui".

I am really sorry for bothering, I am completely new to anaconda and phyton, I only know a tiny bit in matlab, we are using phy on another computers in the lab, we never had this problem before. I am not sure what I am doing wrong.

jaredluckas, 2019-07-25:
Yeah I was having this problem at first for phy2 as well. I found that you will have to download and install git from here -> https://git-scm.com/
Then, you have to add the git.exe (which you will find wherever git has installed onto your computer, for example mine was in `C:\Users\jluckas\AppData\Local\Programs\Git\bin\git.exe`) into your PATH environment variable.
If your problem was the same as mine, you couldn't get the last pip dependency phylib. You can fix this by using `pip install git+https://github.com/cortex-lab/phylib.git` in the phy2 environment once you've installed git and added it to your PATH.

Yliew6-gt, 2020-04-15:
> This is with phy1, right? Could you try with phy2 in a new conda environment?

Hi, I have this exact same issue when running phy1. I have installed phy2 on my computer but I wanted to view some old dataset that I have previously curated with phy1. However the template-gui will not open properly and I got about identical error posted above. Is there a way to use the phy1 while having phy2 installed? Thanks.

[TypeError: sequence item 0: expected str instance, VertexShader found]