cortex-lab/phy #860: Issues running on macOS Mojave (10.14.5)

afergu, 2019-07-22:
I am running the following:
macOS 10.14.5 Mojave
conda 4.6.14

I am installing phy (not phy2). I can successfully create the environment using 

`conda env create -f path/to/environment.yml`

but when I try run `phy template-gui params.py` or even `phy --debug`

-bash: phy: command not found

rossant, 2019-07-23:
do you do `conda activate phy` before running the command?

afergu, 2019-07-23:
Yes, and it changes my environment to (phy) i.e. I see (phy) before the name/path of running in my terminal. But it cannot find the command phy after I activate it.

rossant, 2019-07-23:
once phy is activated, try to do `pip install git+https://github.com/cortex-lab/phy.git` and `pip install git+https://github.com/kwikteam/phy-contrib.git`

afergu, 2019-07-24:
Okay I did that, and now when I run anything with phy (debug or the template) I get an error "ImportError: No module named phylib"

Specifically, I get 
`(phy) Alexandras-iMac:path/to/params/file alex$ phy template-gui params.py
Traceback (most recent call last):
  File "/Users/alex/miniconda3/envs/phy/bin/phy", line 6, in <module>
    from phy.apps import phycli
  File "/Users/alex/miniconda3/envs/phy/lib/python3.5/site-packages/phy/__init__.py", line 18, in <module>
    from phylib.utils import Bunch
ImportError: No module named 'phylib'
`

rossant, 2019-07-24:
ah, it incorrectly installed phy2 instead of phy1. Perhaps try this:
`pip install git+https://github.com/cortex-lab/phy.git@master --upgrade`

afergu, 2019-07-24:
Before you said the previous thing: 

I followed some of these instructions from another issue https://github.com/cortex-lab/phy/issues/843 

I ran `pip install git+https://github.com/cortex-lab/phylib.git` and `python -c "import phylib"`

Then when I run phy I get the error `Import Error: No module named colorcet`

So I did `pip install colorcet `which worked corrrectly. 

Then when I run phy I get the error `ImportError: No module named 'OpenGL'`
I tried versions of `pip install Open GL` `pip install "OpenGL"` and I get the error `  ERROR: Could not find a version that satisfies the requirement OpenGL (from versions: none)
ERROR: No matching distribution found for OpenGL`

I tried to run `pip install `OpenGL` and it made my terminal hang. 

I reopened terminal and ran your above suggestion `pip install git+https://github.com/cortex-lab/phy.git@master --upgrade`

Then I tried `phy --debug` and I get `ModuleNotFoundError: No module named 'numpy'`

afergu, 2019-07-24:
I continued to get module errors for numpy, click, traitlets. I used `pip install` for all of them. 

`phy --debug` seems to find phy

But when I run `phy template-gui params.py` I get the same error `ImportError: No module named 'OpenGL`

rossant, 2019-07-24:
I don't understand why the installation instructions don't work for you.
You'll find the list of dependencies here: https://github.com/cortex-lab/phy/blob/dev/requirements.txt
For the OpenGL package, you should do `pip install pyopengl`

rossant, 2019-07-26:
I've put updated installation instructions here: https://phy.readthedocs.io/en/latest/installation/
Let me know if that works better...

rossant, 2019-08-08:
closing, please reopen if this is still not fixed

closed 2019-08-08 by rossant