cortex-lab/phy #862: Autoscaling in TraceView

yagui, 2019-07-24:
It would be good to be able to disable autoscaling in traceview. When you are navigation trough recordings, sometime a big amplitude artifact make everything super small so you need to rescale, but then when you pass the big amplitude artifact everything is to big so you need to rescale again.

rossant, 2019-07-24:
+1
something that could also be useful would be to use .1/.99 quantiles or something for better robustness to outliers when autocomputing the scaling

rossant, 2019-07-26:
done

closed 2019-07-26 by rossant

yagui, 2019-07-26:
right now is working as I wanted, which is great. Thanks!
Although now I can't go back to the autoscale version. The toggle option is not working for me.
I already tried removing .phy folder in the project folder and ~/.phy/TemplateGUI/state.json and it didn't help.
I prefer this non autoscale version but just so you know, this toggle is not working.

rossant, 2019-07-26:
do you mean that when you toggle the autoscale, and then move around in the trace view, the view is no longer scaling automatically?

yagui, 2019-07-26:
yes

rossant, 2019-08-08:
I couldn't find the problem and it seems to work here. Please feel free to open a new issue with a specific failure example (for example with screenshots) if needed. Thanks