cortex-lab/phy #863: Can't see overlapping traces in WaveformView

rglanz, 2019-07-24:
I've had an issue with the WaveformView freezing for a while now. 

Now, there are no traces shown, and all I can see is a waveform template for the channel. The same template is frozen on every channel.

rglanz, 2019-07-24:
Actually, it looks like it is only happening on the first three channels.

rossant, 2019-07-25:
is it a duplicate of #848 ?

rglanz, 2019-07-25:
It seems similar. Installing PyQtWebEngine did not resolve it. 

Is phy_config.py automatically generated in Windows? If so, I'm having trouble finding it.

rglanz, 2019-07-25:
After a new sort in kilosort and another launch of Phy, it seems to be working correctly now. I'm going to close the issue and assume your plug-in patch is what did the trick.

Thanks!

closed 2019-07-25 by rglanz