cortex-lab/phy #864: "Wrong" labeling of channels after kilosorting?

facazevedo, 2019-07-25:
Hi,

I am using kilosort2 + phy TemplateGUI v2.0a1 + phylib v2.0a1. My data comes from a linear probe with 16 channels. After sorting with Kilosort2, 3 channels are excluded. The resulting channel_map is [0; 1; 5; 6; 7; 8; 9; 10; 11; 12; 13; 14; 15]. Those channels are also saved in the channel_map.npy. However, when I open phy2, the numbering in the waveform view goes from 0 to 12, instead of displaying the true channel numbers that remained (image attached). Would it be possible to have the original labels of the channels displayed?

Best.

[image]

rossant, 2019-07-26:
done!

closed 2019-07-26 by rossant

facazevedo, 2019-07-26:
Thanks a lot!