cortex-lab/phy #865: PermissionError when memmap-ing data file

brendanhasz, 2019-07-26:
Hi!

First off - great piece of software.  I've used Phy 1.x extensively and it's fantastic.

I'm having trouble loading data with the dev version of Phy (as of https://github.com/cortex-lab/phy/commit/c81f26097dd651832ccc5d260912836df59756b5) w/ the most recent version of phylib (https://github.com/cortex-lab/phylib/commit/3d21c75839290842b976dec07192a1a734e57bed).  I'm running Windows 7 with Anaconda 2019.07 and Python 3.7.

When I try to start phy2 via:

```bash
cd Path\To\My\Data\
activate phy2
phy template-gui params.py
```

I get a PermissionError:

```
Traceback (most recent call last):
  File "c:\program files\anaconda3\envs\phy2\lib\runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "c:\program files\anaconda3\envs\phy2\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "C:\Program Files\Anaconda3\envs\phy2\Scripts\phy.exe\__main__.py", line9, in <module>
    sys.exit(phycli())
  File "c:\program files\anaconda3\envs\phy2\lib\site-packages\click\core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "c:\program files\anaconda3\envs\phy2\lib\site-packages\click\core.py", line 717, in main
    rv = self.invoke(ctx)
  File "c:\program files\anaconda3\envs\phy2\lib\site-packages\click\core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\program files\anaconda3\envs\phy2\lib\site-packages\click\core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\program files\anaconda3\envs\phy2\lib\site-packages\click\core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "c:\program files\anaconda3\envs\phy2\lib\site-packages\click\decorators.py", line 17, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "c:\program files\anaconda3\envs\phy2\lib\site-packages\phy\apps\__init__.py", line 137, in cli_template_gui
    template_gui(params_path, **kwargs)
  File "c:\program files\anaconda3\envs\phy2\lib\site-packages\phy\apps\template\gui.py", line 187, in template_gui
    controller = TemplateController(**get_template_params(params_path), **kwargs)
  File "c:\program files\anaconda3\envs\phy2\lib\site-packages\phy\apps\base.py", line 692, in __init__
    self.model = self._create_model(dir_path=dir_path, **kwargs) if model is None else model
  File "c:\program files\anaconda3\envs\phy2\lib\site-packages\phy\apps\template\gui.py", line 84, in _create_model
    return TemplateModel(dir_path=dir_path, **kwargs)
  File "c:\program files\anaconda3\envs\phy2\lib\site-packages\phylib\io\model.py", line 248, in __init__
    self._load_data()
  File "c:\program files\anaconda3\envs\phy2\lib\site-packages\phylib\io\model.py", line 316, in _load_data
    self.traces = self._load_traces(self.channel_mapping)
  File "c:\program files\anaconda3\envs\phy2\lib\site-packages\phylib\io\model.py", line 449, in _load_traces
    for path in paths]
  File "c:\program files\anaconda3\envs\phy2\lib\site-packages\phylib\io\model.py", line 449, in <listcomp>
    for path in paths]
  File "c:\program files\anaconda3\envs\phy2\lib\site-packages\phylib\io\model.py", line 105, in load_raw_data
    return _dat_to_traces(path, n_channels=n_channels_dat, dtype=dtype, offset=offset)
  File "c:\program files\anaconda3\envs\phy2\lib\site-packages\phylib\io\model.py", line 88, in _dat_to_traces
    return np.memmap(dat_path, dtype=dtype, shape=(n_samples, n_channels), offset=offset)
  File "c:\program files\anaconda3\envs\phy2\lib\site-packages\numpy\core\memmap.py", line 275, in __new__
    self.filename = filename.resolve()
  File "c:\program files\anaconda3\envs\phy2\lib\pathlib.py", line 1144, in resolve
    s = self._flavour.resolve(self, strict=strict)
  File "c:\program files\anaconda3\envs\phy2\lib\pathlib.py", line 201, in resolve
    s = self._ext_to_normal(_getfinalpathname(s))

PermissionError: [WinError 32] The process cannot access the file because it is
being used by another process: 'F:\\Path\\To\\My\\Data\\File.dat'
```

However, this doesn't seem to be due to another non-phy process using the file.  I.e., in a separate Python process both

```python
with open('F:\\Path\\To\\My\\Data\\File.dat', 'rb') as fid:
    foo = fid.read(1000)
``` 

and

```python
foo = np.memmap('F:\\Path\\To\\My\\Data\\File.dat', dtype=np.int16, shape=(1000,32), offset=0)
foo[:, :]
```

work without throwing PermissionErrors.

Also, running Phy 1.0.9 starts up and loads the data without any issues.

I'd installed the dev version as per your [instructions on ReadTheDocs](https://phy.readthedocs.io/en/latest/installation/):

```bash
conda create -n phy2 python pip numpy matplotlib scipy h5py pyqt cython -y
conda activate phy2
pip install colorcet pyopengl qtconsole requests traitlets tqdm joblib click mkdocs PyQtWebEngine
pip install git+https://github.com/cortex-lab/phy.git@dev
pip install git+https://github.com/cortex-lab/phylib.git
```

Any help or advice would be much appreciated.  Also apologies if this should have been a phylib issue instead of a phy issue.  Again, thanks for the great software!

Cheers,

Brendan

angelonc, 2019-07-30:
I'm getting this permission error as well on a Windows 7 computer with similar builds of phy and conda. It works fine on a separate Windows 10 computer

rossant, 2019-08-01:
It looks like phy 2 doesn't work on windows 7, but works on windows 10. I'll try to set up a windows 7 virtual machine to test

rossant, 2019-08-06:
It should be fixed with the latest version of phylib, please try again!

angelonc, 2019-08-06:
Reinstalling with the new version fixed it for me, thanks!

closed 2019-08-06 by rossant

brendanhasz, 2019-08-06:
Works for me too now, thanks so much!