cortex-lab/phy #866: Mismatch between channel and cluster IDs?

facazevedo, 2019-07-29:
Hi Cyrille,

1) How does Phy2 determine the channel number of a given cluster displayed under ClusterView? I thought that the correct way was to determine the channel which had the biggest template amplitude for each cluster. This seemed to be working in Phy1 but not in Phy2. An adaptation of my matlab code goes below. 

2) Under clusterView, the best channel for a given cluster seems to be mismatched. In the attached picture, you can see that even though channel 2 wasn't include in this analysis, 
clusters 3, 4 and 5 indicate it as the best one.

Thanks!

```Matlab
% Load kilosort2 results
results_dir = 'D:\DataAnalysis\Kilosort2\monkey\ephys_2018_03_05_1452\Results';
channel_map = double(readNPY(fullfile(results_dir,'channel_map.npy')));
templates = double(readNPY(fullfile(results_dir,'templates.npy')));
spike_templates = double(readNPY(fullfile(results_dir,'spike_templates.npy')));

% Preallocate clusters_n_channels array
[n_clusters, ~, n_channels] = size(templates);
clusters_n_channels = nan(n_clusters,2);

% Determine cluster Ids
unique_spike_templates = unique(spike_templates);
clusters_n_channels(unique_spike_templates + 1, 1) = unique_spike_templates;

% Choose the channel which has the biggest template amplitude for each cluster
for i_clusters = 1 : n_clusters
    max_amplitude_channel_ind = 1;
    max_amplitude_template = max(templates(i_clusters, :, max_amplitude_channel_ind)) - min(templates(i_clusters, : ,max_amplitude_channel_ind));
    for i_channels = 2 : n_channels
        temp_max_amplitude_template = max(templates(i_clusters, :, i_channels)) - min(templates(i_clusters, : ,i_channels));
        if temp_max_amplitude_template > max_amplitude_template
            max_amplitude_template = temp_max_amplitude_template;
            max_amplitude_channel_ind = i_channels;
        end
    end
    clusters_n_channels(i_clusters, 2) = channel_map(max_amplitude_channel_ind);
end
```

[image]

rossant, 2019-07-30:
Hi, is this happening with the very latest version of phy and phylib (last updated 4 days ago)?

facazevedo, 2019-07-30:
Hi Cyrille,

yes. I installed evth from scratch yesterday, even Anaconda was freshly installed. Another way to confirm that this is the newer version is that the channels in the WaveformView are labeled according to Kilosort2, as I've asked you in my [previous issue](https://github.com/cortex-lab/phy/issues/864).

Best.

JoseGuzman, 2019-07-31:
Hi,

I'm using the latest developmental branch from the repository. I do the manual curation with phy after sorting with spyking-circus (latest developmental version, 0.8.4).  I can also confirm that 

- **ClusterView** show channel as zero-based form (not the real channel numbering).In my case, adding 16+channel solves the problem (see violet mark).
- **TraceView** show raw traces in the reverse order as in the WaveformView (see yellow mark)

See attached a screenshot.

[image]

v-hofmann, 2019-08-05:
Hi, 

1. The phy internal "id's" are not running from 0 to n_channels -1 in my case (see cluster view in the screenshot). From what i read this should however be the case (or is there some data excluded from visualization?). 

2. The numbers displayed under "channel" in cluster view seems to be the index that points onto a physical recording site that is stored in the "channel_map.npy". However the numbers displayed in waveform view are the converted recording sites (i.e. "channel_map(channel)"). It seems that it would make sense to convert the "channel" numbers similarly for coherence. 

3. in cluster view i get and "amplitude" column containing numbers and an "Amplitude" column that is empty. i am running phy version 2.0a1 on data that was sorted using kilosort 2 (all installed anew 4 days ago) recorded with IMEC probes.

[image]

As under "waveform view" the converted (channel_map(channel#)) are displayed it would maybe be desireable to have the "channel"

rossant, 2019-08-08:
The problem should be at least partly fixed. Please let me know if there are remaining issues here.

JoseGuzman, 2019-08-08:
The TraceView now show  the channels correctly numbered, but both ISIView and FiringRateView do not appear **when selecting one cluster** (see  
[image].

When **selecting two clusters**, both Views appear normal.
[image]

rossant, 2019-08-08:
Thanks @JoseGuzman, should be fixed now..

rossant, 2019-08-08:
closing, please reopen if new issues arise

closed 2019-08-08 by rossant

JoseGuzman, 2019-08-09:
It works fabulously, as usual!

v-hofmann, 2019-08-15:
Thanks Cyrille,

in general there is more agreement now. However some issues remain: 

1. the number that i see under "channel" in ClusterView and the number that is displayed in front of each channel in TraceView are generally in agreement. However there are issues in WaveformView: when i toggle overlap waveforms, the displayed number agrees with what i see in the TraceView. When i toggle the mean waveform (i.e. by pressing "w"  i get some other numbers, which is also true for the 3rd possible view (which i guess shows the original templates created by kilosort?). 

2. i still have a field "amplitude" and a field "Amplitude" in the ClusterView (i.e. 1x capital 1x non-capital). The field "amplitude" contains values (with the exception of a few clusters) and sorts correctly when pressing the "Amplitude" field.

I get the above issues on separate machines with the latest upgrade of phy.

rossant, 2019-08-23:
Thanks @v-hofmann 
(1) should be fixed
I opened a new issue for (2)

JoseGuzman, 2019-09-11:
I will have to reopen the issue for a minor detail. The TraceView channel ordering will not work if phy is called when >1 probes are analyzed simultaneously. This is after using spyking-circus sorted (latest developmental version). 

Attached you can find a screenshot with the most updated developmental version of phy. Channels must be 1-64 ordered in traceView. Note that WaveformView and ProbeView shows the right numeration
[image]