cortex-lab/phy #867: Phy not playing well with spyking circus output

holmescharles, 2019-08-01:
I am sorting spike with spyking circus. I have set up a separate conda enviroment with phy2 per the instructions on the website ().

To try to use phy, I go to the directory where spyking circus puts it's phy-friendly files (e.g., /PATH/TO/DATA/mydata/mydata.GUI) and run "phy template-gui params.py". I get this huge error print out (shown below). I can't make heads or tails of it and am not familiar enough with phy's code to tackle it. Any idea's what's going on? Am I doing something wrong or is this a bug?

```
13:48:52.508 [E] __init__:63          An error has occurred (AssertionError):
Traceback (most recent call last):
  File "/home/chuck/miniconda3/envs/phy2/bin/phy", line 10, in <module>
    sys.exit(phycli())
  File "/home/chuck/miniconda3/envs/phy2/lib/python3.7/site-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/home/chuck/miniconda3/envs/phy2/lib/python3.7/site-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/home/chuck/miniconda3/envs/phy2/lib/python3.7/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/chuck/miniconda3/envs/phy2/lib/python3.7/site-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/chuck/miniconda3/envs/phy2/lib/python3.7/site-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/home/chuck/miniconda3/envs/phy2/lib/python3.7/site-packages/click/decorators.py", line 17, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/chuck/miniconda3/envs/phy2/lib/python3.7/site-packages/phy/apps/__init__.py", line 137, in cli_template_gui
    template_gui(params_path, **kwargs)
  File "/home/chuck/miniconda3/envs/phy2/lib/python3.7/site-packages/phy/apps/template/gui.py", line 187, in template_gui
    controller = TemplateController(**get_template_params(params_path), **kwargs)
  File "/home/chuck/miniconda3/envs/phy2/lib/python3.7/site-packages/phy/apps/base.py", line 692, in __init__
    self.model = self._create_model(dir_path=dir_path, **kwargs) if model is None else model
  File "/home/chuck/miniconda3/envs/phy2/lib/python3.7/site-packages/phy/apps/template/gui.py", line 84, in _create_model
    return TemplateModel(dir_path=dir_path, **kwargs)
  File "/home/chuck/miniconda3/envs/phy2/lib/python3.7/site-packages/phylib/io/model.py", line 261, in __init__
    self._load_data()
  File "/home/chuck/miniconda3/envs/phy2/lib/python3.7/site-packages/phylib/io/model.py", line 312, in _load_data
    self.sparse_templates = self._load_templates()
  File "/home/chuck/miniconda3/envs/phy2/lib/python3.7/site-packages/phylib/io/model.py", line 552, in _load_templates
    assert cols.shape == (n_templates, n_channels_loc)
AssertionError
```

yger, 2019-08-01:
In theory, export should work.. Can you try to export again, from
spyking-circus, but with the option sparse_export set to False in the
[converting] section (if this is not already the case)?

holmescharles, 2019-08-01:
Did as you suggested. Now I am getting a new error (see below). I also tried setting export_pcs from some to all, but the same error occurs.

```
15:19:31.014 [E] __init__:63          An error has occurred (ValueError): axes don't match array
Traceback (most recent call last):
  File "/home/chuck/miniconda3/envs/phy2/bin/phy", line 10, in <module>
    sys.exit(phycli())
  File "/home/chuck/miniconda3/envs/phy2/lib/python3.7/site-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/home/chuck/miniconda3/envs/phy2/lib/python3.7/site-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/home/chuck/miniconda3/envs/phy2/lib/python3.7/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/chuck/miniconda3/envs/phy2/lib/python3.7/site-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/chuck/miniconda3/envs/phy2/lib/python3.7/site-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/home/chuck/miniconda3/envs/phy2/lib/python3.7/site-packages/click/decorators.py", line 17, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/chuck/miniconda3/envs/phy2/lib/python3.7/site-packages/phy/apps/__init__.py", line 137, in cli_template_gui
    template_gui(params_path, **kwargs)
  File "/home/chuck/miniconda3/envs/phy2/lib/python3.7/site-packages/phy/apps/template/gui.py", line 187, in template_gui
    controller = TemplateController(**get_template_params(params_path), **kwargs)
  File "/home/chuck/miniconda3/envs/phy2/lib/python3.7/site-packages/phy/apps/base.py", line 692, in __init__
    self.model = self._create_model(dir_path=dir_path, **kwargs) if model is None else model
  File "/home/chuck/miniconda3/envs/phy2/lib/python3.7/site-packages/phy/apps/template/gui.py", line 84, in _create_model
    return TemplateModel(dir_path=dir_path, **kwargs)
  File "/home/chuck/miniconda3/envs/phy2/lib/python3.7/site-packages/phylib/io/model.py", line 261, in __init__
    self._load_data()
  File "/home/chuck/miniconda3/envs/phy2/lib/python3.7/site-packages/phylib/io/model.py", line 347, in _load_data
    self.sparse_features = self._load_features()
  File "/home/chuck/miniconda3/envs/phy2/lib/python3.7/site-packages/phylib/io/model.py", line 598, in _load_features
    self._find_path('pc_features.npy'), mmap_mode='r').transpose((0, 2, 1))
ValueError: axes don't match array
```

yger, 2019-08-02:
Can you send me the .GUI folder? Or if not too big the data/params/probe
files? Do you have single channel templates?

rossant, 2019-08-02:
Also interested, if you could send it to me as well

holmescharles, 2019-08-02:
I've made a temp repo with the files (not sure that was the best way to do this) and invited you both as collaborators.

Thanks!

holmescharles, 2019-08-02:
Whoops. I had uploaded the .GUI folder with spare_export=True. Just switched it to the version with spare_export=False.

Also, I tried installing phy in my spyking-circus conda environment and got the same error. For completeness, I am including the environment yaml files here.

SPYKING-CIRCUS:
```
name: circus
channels:
  - spyking-circus
  - anaconda
  - conda-forge
  - defaults
dependencies:
  - _libgcc_mutex=0.1=main
  - backcall=0.1.0=py36_0
  - blosc=1.17.0=he1b5a44_0
  - ca-certificates=2019.5.15=0
  - certifi=2019.6.16=py36_1
  - click=7.0=py36_0
  - colorama=0.4.1=py36_0
  - cycler=0.10.0=py36_0
  - cython=0.29.13=py36he1b5a44_0
  - dbus=1.13.6=he372182_0
  - decorator=4.4.0=py36_1
  - expat=2.2.5=he1b5a44_1003
  - fontconfig=2.13.1=he4413a7_1000
  - freetype=2.10.0=he983fc9_0
  - gettext=0.19.8.1=hc5be6a0_1002
  - glib=2.58.3=h6f030ca_1002
  - gst-plugins-base=1.14.5=h0935bb2_0
  - gstreamer=1.14.5=h36ae1b5_0
  - h5py=2.9.0=nompi_py36hf008753_1102
  - hdf5=1.10.4=nompi_h3c11f04_1106
  - icu=58.2=hf484d3e_1000
  - ipykernel=5.1.1=py36h39e3cac_0
  - ipython=7.7.0=py36h39e3cac_0
  - ipython_genutils=0.2.0=py36_0
  - jedi=0.13.3=py36_0
  - joblib=0.13.2=py36_0
  - jpeg=9c=h14c3975_1001
  - jupyter_client=5.3.1=py_0
  - jupyter_core=4.5.0=py_0
  - kiwisolver=1.1.0=py36hc9558a2_0
  - libblas=3.8.0=10_openblas
  - libcblas=3.8.0=10_openblas
  - libedit=3.1.20181209=hc058e9b_0
  - libffi=3.2.1=hd88cf55_4
  - libgcc=7.2.0=h69d50b8_2
  - libgcc-ng=9.1.0=hdf63c60_0
  - libgfortran-ng=7.3.0=hdf63c60_0
  - libiconv=1.15=h516909a_1005
  - liblapack=3.8.0=10_openblas
  - libopenblas=0.3.6=h6e990d7_5
  - libpng=1.6.37=hed695b0_0
  - libsodium=1.0.16=h1bed415_0
  - libstdcxx-ng=9.1.0=hdf63c60_0
  - libuuid=2.32.1=h14c3975_1000
  - libxcb=1.13=h14c3975_1002
  - libxml2=2.9.9=h13577e0_2
  - matplotlib=3.1.0=py36h5429711_0
  - mpi=1.0=mpich
  - mpi4py=3.0.1=py36hf046da1_0
  - mpich=3.2.1=hc99cbb1_1013
  - ncurses=6.1=he6710b0_1
  - numpy=1.16.4=py36h95a1406_0
  - openblas=0.3.6=h6e990d7_5
  - openssl=1.1.1c=h7b6447c_1
  - pandas=0.25.0=py36hb3f55d8_0
  - parso=0.5.0=py_0
  - patsy=0.5.1=py_0
[... 142 lines cut ...]
    - mkdocs==1.0.4
    - param==1.9.1
    - parso==0.5.1
    - pexpect==4.7.0
    - phy==2.0a1
    - phylib==2.0a1
    - pickleshare==0.7.5
    - prompt-toolkit==2.0.9
    - ptyprocess==0.6.0
    - pyct==0.4.6
    - pygments==2.4.2
    - pyopengl==3.1.0
    - pyqt5==5.13.0
    - pyqt5-sip==4.19.18
    - pyqtwebengine==5.13.0
    - pyyaml==5.1.2
    - pyzmq==18.0.2
    - qtconsole==4.5.2
    - requests==2.22.0
    - tqdm==4.32.2
    - traitlets==4.3.2
    - urllib3==1.25.3
    - wcwidth==0.1.7
prefix: /home/chuck/miniconda3/envs/phy2
```

rossant, 2019-08-08:
It should now work with the latest version of phylib and phy. It looks like you only have 1 PC? That was the issue.

closed 2019-08-08 by rossant