cortex-lab/phy #868: Feature Suggestions and Tweaks

mswallac, 2019-08-08:
Hey Cyrille,

We've been continuing to work on phy and we've solicited suggestions for changes from a number of other users.  Below are some changes that we would love for you to incorporate into the main branch or implement as you see fit.  There seems to be a consensus on the following suggestions on our end:

First a couple things which we've prototyped implementations of:

1. Normalization of templates before scaling in TemplateView.

	Currently, the templates (which start off arbitrarily scaled) are then multiplied twice by amplitude. We think it makes more sense that each template is scaled to the mean waveform amplitude.  We fix this with 23e15d5cef0e360a9ba2c3673ca04006d25e699c.

2. Shift+click on TemplateView to select multiple clusters.

	Done with 735bc9bd3f7b9db3bac99814cc34448d8416a09c.

Next are some issues and additions relating to Amplitude view.  These changes ensure apples-to-apples comparisons between clusters.

1. Raw amplitude should show the maximum - minimum on the principle channel of the first selected cluster (by default at least, see #3 below) for all selected clusters, not on the principle channel of each cluster,.  If amplitude for a cluster is not available on the principle channel of the first selected cluster (because its far away from the first selected cluster), plot zeros. 

```python
    def get_spike_raw_amplitudes(self, spike_ids, channel_ids=None, **kwargs):
        """Return the maximum amplitude of the raw waveforms across all channels."""
        # WARNING: extracting raw waveforms is long!
        waveforms = self.model.get_waveforms(spike_ids, channel_ids)
        if waveforms is None:
            return
        assert waveforms.ndim == 3  # shape: (n_spikes, n_samples, n_channels_loc)
        return (waveforms.max(axis=1) - waveforms.min(axis=1)).max(axis=1)
```

In this function it seems that the raw amplitude is just the largest maximum - minimum considering all channels.

2. Template projections should be the projection of each spike shown onto the first selected cluster's template, not projections onto its own clusters template

3. It would be good to be able to choose the channel from which AmplitudeView draws data (at least for 'raw' and 'feature' modes) in the same way that FeatureView responds to a ctrl+click on the WaveformView. That is, raw amplitudes/feature values would be calculated from the clicked channel instead of the principle channel of the first selected cluster. There would be no effect on the template amplitudes since there is only a single template projection (covering many channels) associated with each spike.

Lastly, various issues/requests raised by others:

1. Allow customization of the columns displayed in ClusterView (ordering, which to show, etc.)

2. Define the length of the displayed waveform in the WaveformView (currently a bit too long). 

3. Increase default font size from 4 to 6 as in 82cadb79cf49170b4d49008e5d9b91c6ca9dc44d.

rossant, 2019-08-08:
Great suggestions! Could you open pull requests for the changes you've already implemented?

> 2\. It would be good to be able to choose the channel from which AmplitudeView draws data

That should already be implemented with alt+click in the feature view, does it work for you and does it do the right thing?

I'll do the rest

rossant, 2019-08-09:
> 1. Raw amplitude should show the maximum - minimum on the principle channel of the first selected cluster (by default at least, see #3 below) for all selected clusters, not on the principle channel of each cluster,.

I think that's already the case, see https://github.com/cortex-lab/phy/blob/dev/phy/apps/base.py#L1095

> 1. Template projections should be the projection of each spike shown onto the first selected cluster's template, not projections onto its own clusters template

Could you clarify what you mean exactly here?

> * Allow customization of the columns displayed in ClusterView (ordering, which to show, etc.)

Good idea! done here as a plugin example : https://phy.readthedocs.io/en/latest/plugins/#customizing-the-columns-of-the-cluster-view (requires the latest dev version of phy)

> 2\. Define the length of the displayed waveform in the WaveformView (currently a bit too long).

The length of the template is used for the length of the waveforms. Changing that would require significant work as the assumption that templates and waveforms have the same length is used in multiple places in the code. Perhaps you could try changing the number of samples in the templates array, either in KS2, or directly in the file (untested).

mswallac, 2019-08-09:
I'll open a pull request for the changes I've implemented and look into the other things you've mentioned now. I will post a detailed response in this thread.

mswallac, 2019-08-09:
> > 2. It would be good to be able to choose the channel from which AmplitudeView draws data
> 
> That should already be implemented with alt+click in the feature view, does it work for you and does it do the right thing?

Ah sorry about that, I didn't realize that it was a click on FeatureView. Although alt+click on FeatureView does work for changing what is displayed for 'feature' mode, it does not affect the 'raw' mode. 

Currently, in order to change which channel features are extracted from I would ctrl+click on WaveformView. Then I would alt+click on FeatureView and the AmplitudeView will then show the feature vs. time view clicked.

Ideally, one would be able to have AmplitudeView change which channel is used to calculate 'raw' or 'feature' values from a click on WaveformView, rather than only being able to determine which 'feature' values to show.

> > Raw amplitude should show the maximum - minimum on the principle channel of the first selected cluster (by default at least, see # 3 below) for all selected clusters, not on the principle channel of each cluster.
> 
> I think that's already the case, see https://github.com/cortex-lab/phy/blob/dev/phy/apps/base.py#L1095

Although the right channels are being passed to the below function, it seems that this function will always do maximum - minimum for all channels, and then take the largest of those values, unless I'm misunderstanding the function.

```python
    def get_spike_raw_amplitudes(self, spike_ids, channel_ids=None, **kwargs):
        """Return the maximum amplitude of the raw waveforms across all channels."""
        # WARNING: extracting raw waveforms is long!
        waveforms = self.model.get_waveforms(spike_ids, channel_ids)
        if waveforms is None:
            return
        assert waveforms.ndim == 3  # shape: (n_spikes, n_samples, n_channels_loc)
        ##########################################################
        # The below line seeems to have the behavior I'm talking about
        return (waveforms.max(axis=1) - waveforms.min(axis=1)).max(axis=1)
        ##########################################################
```

I belive that this commit 38f2e5332a73cc7ab9e43aa160e8c728f2f41b26 would have the desired effect. In this implementation I also make it so that if a cluster doesn't share the channel which is being plotted, zeros will be plotted for that cluster.

> > Template projections should be the projection of each spike shown onto the first selected cluster's template, not projections onto its own clusters template
> 
> Could you clarify what you mean exactly here?

```python

    def get_template_amplitude(self, template_id):
        """Return the maximum amplitude of a template's waveforms across all channels."""
        waveforms = self.model.get_template(template_id).template
        assert waveforms.ndim == 2  # shape: (n_samples, n_channels)
        return (waveforms.max(axis=0) - waveforms.min(axis=0)).max()

    def get_spike_template_amplitudes(self, spike_ids, **kwargs):
        """Return the template amplitudes multiplied by the spike's amplitude."""
        amplitudes = self.model.amplitudes[spike_ids]
        # Spike-template assignments.
        spike_templates = self.model.spike_templates[spike_ids]
        # Find the template amplitudes of all templates appearing in spike_templates.
        unique_template_ids = np.unique(spike_templates)
        # Create an array with the template amplitudes.
        template_amplitudes = np.array(
            [self.get_template_amplitude(tid) for tid in unique_template_ids])
        # Get the template amplitude of every spike.
        spike_templates_rel = _index_of(spike_templates, unique_template_ids)
        assert spike_templates_rel.shape == amplitudes.shape
        # Multiply that by the spike amplitude.
        return template_amplitudes[spike_templates_rel] * amplitudes

```

In the above code the template of each spike is stored in spike_templates (this is always the template of the cluster which these spikes belong to currently). From this the highest maximum - minimum across all channels is calculated for the template in the same way as it is for 'raw' as I mentioned above and then multiplied with the amplitude of that spike. 

This should instead be the dot product (projection) of each spike waveform onto the template of the first cluster. I believe this data is already available via model.get_template_features(). I've implemented this here: f76466a8e222ea1f8e092a86af0b2845b6d7e953

Given that this relies on the data generated by Kilosort and that data is only calculated for the 32 most similar clusters, any clusters not among the 32 most similar will display all zeros in amplitude view for 'template' with this change.

> > Allow customization of the columns displayed in ClusterView (ordering, which to show, etc.)
> 
> Good idea! done here as a plugin example : https://phy.readthedocs.io/en/latest/plugins/#customizing-the-columns-of-the-cluster-view (requires the latest dev version of phy)

Awesome, thanks!

> The length of the template is used for the length of the waveforms. Changing that would require significant work as the assumption that templates and waveforms have the same length is used in [...]

Ok, I'll look into that, thanks!

rossant, 2019-08-11:
Thanks for the explanations and for your code propositions! So all of your suggestions should be implemented now in the dev branch (I merged your branch locally before making some changes so I had to close the pull request). I decided to have the `template` amplitude showing just the numbers saved in `amplitudes.npy` with no further modifications, and adding a new amplitude type named `feature_template` with your proposition (the dot product).

closed 2019-08-11 by rossant

mswallac, 2019-08-12:
Awesome, thanks Cyrille!