cortex-lab/phy #870: Name Error : name 'giverandomname' is not defined

JianguangNi, 2019-08-09:
Hello,

I failed to launch the template-gui (Win10; Anaconda3, Python 3.7.3) after installing the Phy2(2.0a1).  In the same machine, I have installed Klusta/phy Kwik-gui under miniconda3 in a different environment, which worked well. 

Do you have any insights to solve this "NameError" issue?   
The params.py file for testing is from SpykingCircus.  

Thanks!
Best, Jianguang

[image]
 
> phy template-gui params.py

[31m12:07:49.648 [E] __init__:62          An error has occurred (NameError): name 'giverandomname' is not defined
Traceback (most recent call last):
  File "c:\users\jni03\appdata\local\continuum\anaconda3\envs\phy2ana\lib\runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "c:\users\jni03\appdata\local\continuum\anaconda3\envs\phy2ana\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "C:\Users\jni03\AppData\Local\Continuum\anaconda3\envs\phy2Ana\Scripts\phy.exe\__main__.py", line 9, in <module>
    sys.exit(phycli())
  File "c:\users\jni03\appdata\local\continuum\anaconda3\envs\phy2ana\lib\site-packages\click\core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "c:\users\jni03\appdata\local\continuum\anaconda3\envs\phy2ana\lib\site-packages\click\core.py", line 717, in main
    rv = self.invoke(ctx)
  File "c:\users\jni03\appdata\local\continuum\anaconda3\envs\phy2ana\lib\site-packages\click\core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\users\jni03\appdata\local\continuum\anaconda3\envs\phy2ana\lib\site-packages\click\core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\users\jni03\appdata\local\continuum\anaconda3\envs\phy2ana\lib\site-packages\click\core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "c:\users\jni03\appdata\local\continuum\anaconda3\envs\phy2ana\lib\site-packages\click\decorators.py", line 17, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "c:\users\jni03\appdata\local\continuum\anaconda3\envs\phy2ana\lib\site-packages\phy\apps\__init__.py", line 136, in cli_template_gui
    template_gui(params_path, **kwargs)
  File "c:\users\jni03\appdata\local\continuum\anaconda3\envs\phy2ana\lib\site-packages\phy\apps\template\gui.py", line 187, in template_gui
    controller = TemplateController(**get_template_params(params_path), **kwargs)
  File "c:\users\jni03\appdata\local\continuum\anaconda3\envs\phy2ana\lib\site-packages\phylib\io\model.py", line 952, in get_template_params
    params = read_python(params_path)
  File "c:\users\jni03\appdata\local\continuum\anaconda3\envs\phy2ana\lib\site-packages\phylib\utils\_misc.py", line 173, in read_python
    exec(contents, {}, metadata)
  File "<string>", line 1, in <module>
NameError: name 'giverandomname' is not defined
[0m

rossant, 2019-08-09:
could you post the params.py file?

JianguangNi, 2019-08-09:
------------ params.py
dat_path = giverandomname.dat
n_channels_dat = 32
n_features_per_channel = 5
dtype = r">i2"
offset = 1024
sample_rate = 30000.0
dir_path = r"C: \path-to-the-data\Continuous_Data.GUI"
hp_filtered = True

rossant, 2019-08-09:
You need to replace the first line with the actual path to the raw data file, with quotes, eg:

```python
dat_path = r"C:\path-to-the-data\Continuous_Data.GUI\giverandomname.dat"
```

The exact path depends on where you've put the data files on your hard drive.

JianguangNi, 2019-08-09:
Hey Cyrille,
I actually don't have the raw data. Only the params.py file from other people. I will get the raw data and test it again.

yger, 2019-08-10:
Indeed, looks like this is a big from spkykng circus. We write a fake name (giverandomname.dat) when the data file is not supported by phy, but the first line should be with quotes...

closed 2019-08-23 by rossant