cortex-lab/phy #871: unsorted clusters missing in cluster_group.tsv file

HiroyukiKato7, 2019-08-10:
In the most recent phy2 from dev brunch, saving the data results in a cluster_group.tsv file which skips all the unsorted clusters.  It includes good, mua, and noise clusters, but unsorted ones are completely missing.

It was not like this in the previous version that I downloaded a month ago, so this must be a result of a recent change.  It is causing some trouble in my analysis, but is this on purpose?

rossant, 2019-08-10:
Yes, the idea is that only sorted clusters need to be saved in this file. How is it breaking your analysis?

HiroyukiKato7, 2019-08-10:
I understand.  I can modify my codes to handle this. 

The reason was this: I calculate quality measures (ISI, contamination ratio, etc) before running Phy and use them to help deciding good units.  For the units that I split/merge, I leave them unsorted temporarily and recalculate quality measures before making final decisions for good units.  My original code was referring to cluster_group.tsv to obtain cluster IDs, and I got confused because now it was skipping unsorted ones.  I'll make a change on my side, so it's not a problem.

Thanks again for making this code available!

rossant, 2019-08-11:
Thanks for the explanations! A couple of things you might want to know:

* There should be a `cluster_info.tsv` file automatically saved, with all clusters and all columns shown in the cluster view. Perhaps you could use that to obain the cluster IDs.

* You'll find plugin examples showing how to add custom cluster metrics in phy, which are automatically recomputed when splitting/merging, and are automatically saved in `cluster_info.tsv`

closed 2019-08-11 by rossant