cortex-lab/phy #872: Small Bugs and Tweaks

mswallac, 2019-08-12:
Hey Cyrille,

Here are a few small things I wanted to follow up on from my last post #868. 

1. Error when launching on latest version of Phy

```console
(phy2new) D:\dl59\20181207>phy template-gui params.py
[33m10:46:08.488 [W] model:417            Could not load D:\dl59\20181207\goCue.csv: not enough values to unpack (expected 2, got 1).[0m
[31m10:46:08.943 [E] __init__:62          An error has occurred (SystemError): <built-in function connectSlotsByName> returned a result with an error set
AttributeError: __getattribute__

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "C:\Users\labadmin\Miniconda3\envs\phy2new\Scripts\phy-script.py", line 11, in <module>
    load_entry_point('phy', 'console_scripts', 'phy')()
  File "C:\Users\labadmin\Miniconda3\envs\phy2new\lib\site-packages\click\core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "C:\Users\labadmin\Miniconda3\envs\phy2new\lib\site-packages\click\core.py", line 717, in main
    rv = self.invoke(ctx)
  File "C:\Users\labadmin\Miniconda3\envs\phy2new\lib\site-packages\click\core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "C:\Users\labadmin\Miniconda3\envs\phy2new\lib\site-packages\click\core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "C:\Users\labadmin\Miniconda3\envs\phy2new\lib\site-packages\click\core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "C:\Users\labadmin\Miniconda3\envs\phy2new\lib\site-packages\click\decorators.py", line 17, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "c:\users\labadmin\desktop\phy\newphy\phy-dev\phy\apps\__init__.py", line 136, in cli_template_gui
    template_gui(params_path, **kwargs)
  File "c:\users\labadmin\desktop\phy\newphy\phy-dev\phy\apps\template\gui.py", line 188, in template_gui
    gui = controller.create_gui()
  File "c:\users\labadmin\desktop\phy\newphy\phy-dev\phy\apps\base.py", line 1340, in create_gui
    **kwargs)
  File "c:\users\labadmin\desktop\phy\newphy\phy-dev\phy\gui\gui.py", line 164, in __init__
    QMetaObject.connectSlotsByName(self)
SystemError: <built-in function connectSlotsByName> returned a result with an error set
[0m
```

I fixed this issue by removing the line: 
```python 
QMetaObject.connectSlotsByName(self) in phy/gui/gui.py, line 164
```
Based on this post: https://stackoverflow.com/questions/2462401/problem-in-understanding-connectslotsbyname-in-pyqt

I'm not very well acquainted with that code, so I'm not sure if this is actually a desirable way to fix this issue.

2. Channel being used to display amplitudes does not automatically update when first cluster selection changes (will show zeros if I switch to some other far away cluster until I ctrl+click on WaveformView again).

3. Alt+click on FeatureView seems to be misbehaving a bit in terms of where the mouse needs to be placed. It will often not work when clicking directly on the desired Feature vs. Time plot, but will work when clicking on the plots above or below that plot.

4. Ctrl+click on WaveformView will always change AmplitudeView back to 'raw' mode. It would be nice if AmplitudeView just retained its current mode. However, this isn't too important and could be left as-is.

Let me know if you'd like me to implement any of these or provide more detailed explanations. 

Cheers!

rossant, 2019-08-13:
Should be all done, please reopen and let me know if I forgot something!

closed 2019-08-13 by rossant

tlstay, 2019-08-13:
Hello,
I just installed Phy yesterday and am getting the same error as mswallac when trying to open the GUI.  Is their suggestion to delete line 164 the correct way to resolve this problem?  I wasn't clear what "please reopen" meant; I tried reinstalling Phy just now but am getting the same error.  Thanks!

rossant, 2019-08-13:
The suggestion to delete that line seems to work and should be live on the dev branch. Could you copy here the traceback of the error?

tlstay, 2019-08-13:
Sure.  Thank you for the quick response.

(phy2) C:\Users\tstay\Box\Trace Stay's Externally Shareable Files\SGL_DATA\testBrain_g2_may have actual data\testBrain_g2_imec0>phy template-gui params.py
 [31m10:51:40.442 [E] __init__:62          An error has occurred (SystemError): <built-in function connectSlotsByName> returned a result with an error set
AttributeError: __getattribute__

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "c:\users\tstay\.conda\envs\phy2\lib\runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "c:\users\tstay\.conda\envs\phy2\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "C:\Users\tstay\.conda\envs\phy2\Scripts\phy.exe\__main__.py", line 9, in <module>
    sys.exit(phycli())
  File "c:\users\tstay\.conda\envs\phy2\lib\site-packages\click\core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "c:\users\tstay\.conda\envs\phy2\lib\site-packages\click\core.py", line 717, in main
    rv = self.invoke(ctx)
  File "c:\users\tstay\.conda\envs\phy2\lib\site-packages\click\core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\users\tstay\.conda\envs\phy2\lib\site-packages\click\core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\users\tstay\.conda\envs\phy2\lib\site-packages\click\core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "c:\users\tstay\.conda\envs\phy2\lib\site-packages\click\decorators.py", line 17, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "c:\users\tstay\.conda\envs\phy2\lib\site-packages\phy\apps\__init__.py", line 136, in cli_template_gui
    template_gui(params_path, **kwargs)
  File "c:\users\tstay\.conda\envs\phy2\lib\site-packages\phy\apps\template\gui.py", line 188, in template_gui
    gui = controller.create_gui()
  File "c:\users\tstay\.conda\envs\phy2\lib\site-packages\phy\apps\base.py", line 1340, in create_gui
    **kwargs)
  File "c:\users\tstay\.conda\envs\phy2\lib\site-packages\phy\gui\gui.py", line 164, in __init__
    QMetaObject.connectSlotsByName(self)
SystemError: <built-in function connectSlotsByName> returned a result with an error set

rossant, 2019-08-13:
It looks like phy has not been properly updated. Could you do:

`pip install --upgrade git+https://github.com/cortex-lab/phy.git`

and try again?

tlstay, 2019-08-13:
That worked, thank you!  Sorry, I didn't realize that there was a difference between 
pip install git+https://github.com/cortex-lab/phy.git@dev
and the --upgrade command.  One note, I did get this message:
Could not copy non-existing default state file c:\users\tstay\.conda\envs\phy2\lib\site-packages\phy\apps\template\static\state.json
but the GUI itself opened up.  So thank you!

mswallac, 2019-08-13:
> Should be all done, please reopen and let me know if I forgot something!

Seems like everything here is resolved. Thanks Cyrille!