cortex-lab/phy #873: How to label only best?

ablot, 2019-08-13:
Labelling a cluster (using "L") will apply the label to both best and the closest match. I have the feeling that it wasn't the case before but maybe I just never noticed.

Typically I select a best, press space a bunch of times and decide at the end if I want to add a label, is there a way to easily label only the best without affecting the closest match?

rossant, 2019-08-13:
The previous behavior might have been unintended as this action should normally act on all currently selected clusters. Perhaps a new action named `label_best` with e.g. shortcut `ctrl+l` would work?

ablot, 2019-08-13:
That would be perfect but that would override the toggle label. A keyboard shortcut for deselecting all similar would also work (`backspace`?). 
I can also live without, for now I do `up`, `down` to go to the previous best and then back to my current best with no similar select (that's just sometimes a bit slow with bigger datasets).

rossant, 2019-08-13:
> A keyboard shortcut for deselecting all similar would also work (`backspace`?).

nice idea! I'll do that

commit 081425b1 (2019-08-13): Add docs. Closes #873
closed 2019-08-13 by rossant