cortex-lab/phy #874: JavaScript error when launching kwik-gui.

fbens47, 2019-08-16:
Hi,

I just moved to phy2 from phy1. The installation went well. But I got this error when I launch phy (version 2.0a1) on MacOS Mojave 10.14.6 (command `phy kwik-gui my_file.kwik`): 
`qt:352               [JS:L1629] Uncaught ReferenceError: Table is not defined`.

Any ideas?

Thanks!

rossant, 2019-08-23:
Do you get a more complete error message by appending the `--debug` option to the `phy` command?

emilyjanedennis, 2019-08-24:
js: Uncaught ReferenceError: Table is not defined
19:54:17.667 [E] qt:352               [JS:L1629] Uncaught ReferenceError: Table is not defined

this is using a known good data set from the kilosort2 workshop
before the reference error, it says: 

19:53:55.824 [D] __init__:68          Start capturing exceptions.
19:53:56.121 [D] model:514            Loading spike clusters.
19:53:56.149 [D] model:468            No channel shank file found.
19:53:56.156 [D] model:543            Loading templates.
19:53:56.170 [D] model:569            Templates are dense.
19:53:56.170 [D] model:575            Loading the whitening matrix.
19:53:56.173 [D] model:582            Loading the inverse of the whitening matrix.
19:53:56.195 [D] model:159            Loading traces at `/Users/emilydennis/Desktop/201908_KilosortWorkshop/example_data/SC026_080819_g0_20min.imec2.ap.bin`.
19:53:56.196 [D] model:611            Loading features.
19:53:56.742 [D] model:626            Features are sparse.
19:53:56.742 [D] model:648            Loading template features.
19:53:56.747 [D] model:658            Template features are sparse.
19:53:56.748 [D] model:413            Load `/Users/emilydennis/Desktop/201908_KilosortWorkshop/example_data/cluster_Amplitude.tsv`.
19:53:56.751 [D] _misc:297            Read /Users/emilydennis/Desktop/201908_KilosortWorkshop/example_data/cluster_Amplitude.tsv.
19:53:56.751 [D] model:413            Load `/Users/emilydennis/Desktop/201908_KilosortWorkshop/example_data/cluster_ContamPct.tsv`.
19:53:56.754 [D] _misc:297            Read /Users/emilydennis/Desktop/201908_KilosortWorkshop/example_data/cluster_ContamPct.tsv.
19:53:56.754 [D] model:413            Load `/Users/emilydennis/Desktop/201908_KilosortWorkshop/example_data/cluster_KSLabel.tsv`.
19:53:56.758 [D] _misc:297            Read /Users/emilydennis/Desktop/201908_KilosortWorkshop/example_data/cluster_KSLabel.tsv.
19:53:56.828 [D] context:100          Initialize joblib cache dir at `/Users/emilydennis/Desktop/201908_KilosortWorkshop/example_data/.phy`.
19:53:56.828 [D] context:101          Reducing the size of the cache if needed.
19:53:56.841 [D] config:31            Load config file `/Users/emilydennis/.phy/phy_config.py`.
19:53:56.842 [D] plugin:145           Loading 0 plugins.
19:53:56.843 [D] context:126          Load memcache for `phy.apps.base.get_mean_spike_raw_amplitudes`.
19:53:56.844 [D] context:126          Load memcache for `phy.apps.base._get_mean_waveforms`.
19:53:56.847 [D] context:126          Load memcache for `phy.apps.base.get_mean_spike_raw_amplitudes`.
19:53:56.848 [D] context:126          Load memcache for `phy.apps.base._get_mean_waveforms`.
19:53:56.849 [D] context:126          Load memcache for `phy.apps.base._get_template_waveforms`.
19:53:56.850 [D] context:126          Load memcache for `phy.apps.base.get_mean_spike_template_amplitudes`.
19:53:56.851 [D] context:126          Load memcache for `phy.apps.base.get_template_counts`.
19:53:56.853 [D] context:126          Load memcache for `phy.apps.base.get_template_for_cluster`.
19:53:56.855 [D] context:126          Load memcache for `phy.apps.template.gui.get_template_amplitude`.
19:53:56.857 [D] context:126          Load memcache for `phy.apps.base.get_cluster_amplitude`.
19:53:56.858 [D] context:126          Load memcache for `phy.apps.base.get_mean_firing_rate`.
19:53:56.858 [D] context:126          Load memcache for `phy.apps.base.get_best_channel`.
19:53:56.860 [D] context:126          Load memcache for `phy.apps.template.gui.get_best_channels`.
19:53:56.866 [D] context:126          Load memcache for `phy.apps.base.get_channel_shank`.
19:53:56.867 [D] context:126          Load memcache for `phy.apps.base.get_probe_depth`.
19:53:56.867 [D] context:126          Load memcache for `phy.apps.base.peak_channel_similarity`.
19:53:56.983 [D] context:185          Save data to `/Users/emilydennis/Desktop/201908_KilosortWorkshop/example_data/.phy/spikes_per_cluster.pkl`.
19:53:57.131 [D] gui:174              Creating GUI.
19:53:57.215 [D] state:46             Load /Users/emilydennis/.phy/TemplateGUI/state.json for GUIState.
19:53:57.216 [D] state:46             Load /Users/emilydennis/Desktop/201908_KilosortWorkshop/example_data/.phy/state.json for GUIState.
19:53:57.595 [D] gui:423              Add view ClusterView to GUI.
19:53:57.601 [D] gui:423              Add view SimilarityView to GUI.
19:53:57.711 [D] gui:423              Add view WaveformView to GUI.
19:53:57.714 [D] base:310             Set state for WaveformView.
19:53:57.761 [D] gui:423              Add view CorrelogramView to GUI.
19:53:57.763 [D] base:310             Set state for CorrelogramView.
19:53:57.793 [D] gui:423              Add view ISIView to GUI.
19:53:57.794 [D] base:310             Set state for ISIView.
19:53:57.812 [D] gui:423              Add view FeatureView to GUI.
19:53:57.814 [D] base:310             Set state for FeatureView.
19:53:57.838 [D] gui:423              Add view AmplitudeView to GUI.
19:53:57.840 [D] base:310             Set state for AmplitudeView.
19:53:57.866 [D] gui:423              Add view FiringRateView to GUI.
19:53:57.867 [D] base:310             Set state for FiringRateView.
19:53:57.936 [D] gui:423              Add view ProbeView to GUI.
19:53:57.938 [D] base:310             Set state for ProbeView.
19:53:57.954 [D] gui:423              Add view TemplateFeatureView to GUI.
19:53:57.956 [D] base:310             Set state for TemplateFeatureView.
19:53:57.958 [D] gui:225              Load the geometry state.

fbens47, 2019-08-27:
Same output as @emilyjanedennis.

wrobison347, 2019-09-09:
I'm also having the same problem!

fbens47, 2020-01-27:
No more error, Qt sends now a weird alert only on MacOS Mojave (but there is no crash): 
`2020-01-27 13:07:45.430 QtWebEngineProcess[16913:945871] _CFURLCopyResourcePropertyValuesAndFlags failed because it was passed an URL which has no scheme`.

Maybe an issue with the Qt version I use. I close this issue.

Flavien

closed 2020-01-27 by fbens47