cortex-lab/phy #875: phy 2.0 template-gui issue running with scientfic linux 7

jenniferColonell, 2019-08-18:
When running with the lastest Anaconda 3 and Scientific Linux 7, I had a problem when trying to launch the GUI. I was able to fix the problem by replacing the pyqt installed by pip with pyqt installed from anaconda ( conda install -c anaconda pyqt) -- that was a result of trial and error.

I'm attaching the error message I get when running with the pip pyqt (that is just following the install instructions). I ran export QT_DEBUG_PLUGINS=1 beforehand to get the details:

[phy2_error.txt](https://github.com/cortex-lab/phy/files/3512615/phy2_error.txt)

rossant, 2019-08-23:
Did you follow the instructions at https://phy.readthedocs.io/en/latest/installation/ ? Normally the anaconda pyqt is the one that should be installed, not the pip one, otherwise it means there's a problem with the installation instructions

jenniferColonell, 2019-08-23:
Yes, I followed those instructions. Looking at the list of packages in the environment, the pyqt installed was from pip, until I installed the anaconda version over it. Thanks!

rossant, 2019-08-23:
Okay, perhaps it was due to the SL7 distribution. I'll reopen if the same issue arises to someone else

closed 2019-08-23 by rossant