cortex-lab/phy #879: sampling rate incorrect

nsteinme, 2019-08-22:
I'm using a neuropixels 1.0 probe with sampling rate 30kHz. The params.py file indeed has: 

```
sample_rate = 30000.
```
However, the gui is plotting the data incorrectly. In this screenshot you can see that the time between two samples of the waveforms is plotted as about 0.01ms, whereas it ought to be 0.033ms. 
[image]

This also explains why I'm seeing much more "dead space" around the neuronal waveform than I'm used to: it's pulling out what it thinks is 1 ms of data, but it is actually about 3.3ms of data. 

[image]

This is an important bug!!

rossant, 2019-08-23:
There are two separate issues here:

1. The axis labels are wrong, I'm working on it.
2. In principe the "dead space" is an unrelated issue: to make the implementation simpler, the number of time samples for waveform extraction is equal to the number of time samples in the template waveforms (exported by KiloSort)

nsteinme, 2019-08-23:
Got it - then let me make a suggestion for a fix: just like you can set the number of bins for CCG, also allow the user to set the number of WF samples to load/display (and the number of channels to load/display!). If these things are already set-able - sorry that I missed it!

rossant, 2019-08-23:
okay, opened #886 for (2), and finishing to fix (1)

closed 2019-08-23 by rossant