cortex-lab/phy #881: amplitude sorting doesn't work?

nsteinme, 2019-08-22:
It's weird because it sort of works: units with generally higher amplitude end up higher in the list, or the reverse - but it is not precise. 
[image]

[image]

rossant, 2019-10-07:
fixed by renaming the duplicate column "amplitude" to "amp"

closed 2019-10-07 by rossant

jaib1, 2019-11-12:
I still have this issue - I installed phy < 1 month ago according to the [readthedocs](https://phy.readthedocs.io/en/latest/installation/) instructions.

rossant, 2019-11-12:
there was a conflict between `amplitude` and `Amplitude` but this should have been fixed?!

jaib1, 2019-11-13:
ah ok, maybe it's because I actually installed phylib directly from pypi instead of github? 
I did 
`pip install phylib` 
instead of 
`pip install git+https://github.com/cortex-lab/phylib.git`
Let me try reinstalling.