cortex-lab/phy #883: Error in the 'WaveformView'

jiangjinxiang, 2019-08-23:
Dear Sir/Madam,
  I met the bug in 'WaveformView' interface as follow:
[image]
[image]
  There's no image about spike waveform in TemplateGUI, please kindly help.

rossant, 2019-08-23:
Could you launch again phy by adding the `--clear-state` command and see if it solved the issue? The scaling of the waveform view might be incorrect and this option would reset it to the default

jiangjinxiang, 2019-08-24:
> Could you launch again phy by adding the `--clear-state` command and see if it solved the issue? The scaling of the waveform view might be incorrect and this option would reset it to the default

Sir, it has been shown that 'It's not an internal or external command' in phy mode while tpying '--clear-state';

rossant, 2019-08-24:
okay so you might have an older version of phy. Try doing:

```
pip install --upgrade git+https://github.com/cortex-lab/phy.git@dev
pip install --upgrade git+https://github.com/cortex-lab/phylib.git
```

jiangjinxiang, 2019-08-24:
> okay so you might have an older version of phy. Try doing:
> 
> ```
> pip install --upgrade git+https://github.com/cortex-lab/phy.git@dev
> pip install --upgrade git+https://github.com/cortex-lab/phylib.git
> ```

Actually, Sir, I have installed the lastest version, and it does not work; and I tried to follow the installation from the 'phy' website (https://phy.readthedocs.io/en/latest/installation/);
I met the notices that there was no 'git' in the python enviroment while doing that conmands (pip install git+https://github.com/cortex-lab/phy.git@dev
pip install git+https://github.com/cortex-lab/phylib.git), so I installed it from 'phy2' enviroment in the Anaconda; Maybe I made some mistakes here?

rossant, 2019-08-24:
If git is not installed, perhaps it could not install the very latest version. Try doing `conda install git` after having activated the conda environment, and try doing the pip commands again

jiangjinxiang, 2019-08-24:
> If git is not installed, perhaps it could not install the very latest version. Try doing `conda install git` after having activated the conda environment, and try doing the pip commands again

Sir, I have followed your suggestions, and there is sitll no the '--clear-state' option in the phy

rossant, 2019-08-24:
What are the output of

```
phy --version
phy template-gui --help
```

jiangjinxiang, 2019-08-24:
> Sir, I have followed your suggestions, and there is sitll no the '--clear-state' option in the phy.
>
> [image]

phy, version 2.0a1

Usage: phy template-gui [OPTIONS] PARAMS_PATH

  Launch the template GUI on a params.py file.

Options:
  --clear-state / --no-clear-state
                                  Clear the GUI state in `~/.phy/` and in
                                  `.phy`.
  --clear-cache / --no-clear-cache
                                  Clear the .phy cache in the data directory.
  --help                          Show this message and exit.

rossant, 2019-08-24:
The clear state option seems to be available actually, so try to launch the GUI as follows: `phy template-gui params.py --clear-state`

jiangjinxiang, 2019-08-24:
> The clear state option seems to be available actually, so try to launch the GUI as follows: `phy template-gui params.py --clear-state`

[image]

[image]

it seem not work,Sir.

rossant, 2019-08-24:
It seemed that this was not the issue: the raw data file is not available. Therefore, the waveforms cannot be displayed. Try to press "w" on the keyboard several times to see if you manage to see the templates

closed 2019-09-23 by rossant