cortex-lab/phy #887: Show time-dependent templates

rossant, 2019-08-26:
Show the time-dependent templates (which would need to be saved into a new file).

@marius10p : could you tell me more about what would be for you the most efficient way of visualizing this.

* New view or existing view?
* How to show time: superimposed templates with a time-dependent color? Or one time-dependent template per column? 
* Show only 1 neuron? Show the N selected neurons? Show all neurons at once?
* Would there need to be a text label (for example the time of the template, or the template index, etc.)?
* Spec of the new file: filename, shape, dense or sparse format, etc.

A hand-drawn sketch might possibly be useful.

marius10p, 2019-09-19:
- I think this view should be kept separate because it is an advanced feature and might be confusing for beginners. 
- Superimposed might emphasize the changes better, so let's try that first. 
- If we do superimposed, then we should show the N selected neurons. This would help emphasize coordinated changes in waveforms. 
- No label if we do superimposed. 
- We (+@nsteinme) have used a PCA compressed representation for time-dependent templates in the Matlab GUI. We use that currently for the template subtraction. The compression happens in lines 342-359 of learnAndSolve8b. It's a little tricky, because what's compressed are actually the already compressed PCA  decomposition of the template (in time and space, like figure 2 of the NIPS paper).  It might be be best if I provide a Python function to reconstruct time-varying templates from the compressed representation? What is the internal Phy representation of templates? Do you want to just add a time-axis for that? I think it should only be done "on-demand" and not precomputed (the matrices are large otherwise).

rossant, 2019-09-23:
Okay for superimposed. Yes it would be great to have a Python function. For each cluster, I would like the templates to have the shape `(n_templates_neuron, n_samples, n_peak_channels)` where:

* `n_templates_neuron` is the number of template waveforms for that template
* `n_samples` is the number of time samples of each template
* `n_peak_channels` is the number of channels on which the template appears

Perhaps we could use a linear colormap for the templates, where each waveform's color would represent the time of that template shape along the recording.

rossant, 2019-10-08:
@marius10p I'll implement this once you send me a Python function that extracts the templates