cortex-lab/phy #888: Time reordering

rossant, 2019-08-26:
Support for different time reorderings.

* For example, there could be a set of special files `spike_times.myreorder1.npy` that would be like the existing `spike_times.npy`, but with different spike times (for example, a reordering of the normal spike times).
* One could switch between these different sets of spike times with a keyboard shortcut.
* Which views should be updated accordingly @marius10p ?
    * [ ] Diagonal subplots in Feature view ?
    * [ ] Correlograms?
    * [ ] Spikes in trace view?
    * [ ] Amplitude view?
    * [ ] Raster view?
    * [ ] Spike attribute views? (these are custom views with 1 value per spike, the x axis is the time, it could be the "true" time or the "custom" time depending on what the user chose)

marius10p, 2019-09-19:
I think just those two are enough.

rossant, 2019-10-08:
@marius10p would it be sufficient to add a single file `spike_reorder.npy` which would contain integers, representing a permutation of the original `spike_times.npy`? Or is there a need for multiple reorderings, and a need for spike times that are completely different from the original ones, rather than just a permutation of the original ones?

rossant, 2019-10-11:
Done: just create a `spike_reorder.npy` file with a permutation of the spike indices. Then, you can use `ctrl+r` to toggle spike reordering in the amplitude view, and in the diagonals of the feature view.

closed 2019-10-11 by rossant

marius10p, 2019-10-12:
Hi Cyrille, this doesn't seem right. We don't need a permutation of the
spike indices, we need new timestamps for each spike. These are not the
same, particularly when the density of spikes is different in different
time or reordered-time epochs.

rossant, 2019-10-12:
No problem, this is fixed. You can now simply add `spike_times_reordered.npy` and toggle it with `ctrl+r`.

czuba, 2020-07-02:
Hey guys,

I wanted to take a look at spike amplitudes in batch-reordered state, but I'm not sure where/how to generate a corresponding `spike_times_reordered.npy` output.  Is there a straightforward point within the normal processing stream (`rezToPhy`?) to just add an extra column to st3 for these reordered timestamps?

Even without creating an alternate `spike_times_reordered.npy` file, toggling `ctrl+r` in Phy _does produce slight data point shifts & histogram bin changes_ in the AmplitudeView, leaving me extra confused as to what/how that was implemented

Thanks!