cortex-lab/phy #889: Trace and waveform filtering

rossant, 2019-08-26:
Support multiple types of time filtering in traces, that would also affect extracted waveforms (in the waveform view) and raw amplitudes in the amplitude view.

@marius10p could you give more information about what filters exactly you would like and how they should be computed (with mathematical formulas or pseudocode)?

marius10p, 2019-09-19:
I like the idea of this affecting the extracted waveforms! Would this option be switchable quickly? Sometimes we want to see the raw data waveforms because filtering can change their shape in funny ways. Switching back and forth to see the change in shape from filtering would be valuable. 

In Kilosort2, we do the following by default:

1) median subtraction across all channels
2) high-pass filtering at 150Hz using a 3-node butterworth filter. The matlab formula is:

[b1, a1] = butter(3, ops.fshigh/ops.fs*2, 'high');
datr = filter(b1, a1, dataRAW);
datr = flipud(datr);
datr = filter(b1, a1, datr);
datr = flipud(datr); 

This is done to avoid Matlab's filtfilt function, which requires double inputs and so will be slow on most GPUs. The compromise is to use a butterworth filter with very few nodes (3).

rossant, 2019-10-11:
Done! I added [a plugin example in the documentation](https://phy.readthedocs.io/en/latest/plugins/#adding-a-custom-raw-data-filter).

closed 2019-10-11 by rossant

410pfeliciano, 2020-01-15:
Hello all,

I remembered switching from raw to filter data in the trace view, but I forgot the shortcut to do that. Alt+R is not working on my desktop.  Is there any other shortcut?

rossant, 2020-01-15:
You should find the action in the view menu