cortex-lab/phy #892: Selecting cluster in TraceView

yagui, 2019-09-05:
When trying to select another cluster (using Alt+s to view all clusters) the cluster selected is in an opposite channel, respect to the middle, from the one intended to select.

For example, if I tried to select a spike on the lower channel (Number 0), it gets selected the closest spike in time, from the upper channel (Last channel). 

Using latest commit 1aae483022cce3d0562437fcfe273b2c2a649da1

rossant, 2019-10-07:
I can't reproduce this bug, could you send me your dataset? Thanks.

yagui, 2019-10-09:
[sim_binary.zip](https://github.com/cortex-lab/phy/files/3707835/sim_binary.zip)
I used [this](https://github.com/kwikteam/phy-data/blob/master/template/sim_binary.dat) data file and it was sorted with spyking-circus.

yagui, 2019-10-09:
It happens to me with every dataset that I work with. Is just that all of them are 5-6GB so I think it is easier to use that as an example. I just downloaded latest version of phy and phylib and keeps happening to me. I can try to make a video if you want.

yagui, 2019-10-10:
Yes, Thanks!

closed 2019-10-10 by yagui