cortex-lab/phy #893: I will have to reopen the issue for a minor detail. The TraceView channel ordering will not work if phy is called when >1 probes are analyzed simultaneously. This is after using spyking-circus sorted (latest developmental version).

JoseGuzman, 2019-09-11:
I will have to reopen the issue for a minor detail. The TraceView channel ordering will not work if phy is called when >1 probes are analyzed simultaneously. This is after using spyking-circus sorted (latest developmental version). 

Attached you can find a screenshot with the most updated developmental version of phy. Channels must be 1-64 ordered in traceView. Note that WaveformView and ProbeView shows the right numeration
[image]

_Originally posted by @JoseGuzman in https://github.com/cortex-lab/phy/issues/866#issuecomment-530259426_

rossant, 2019-10-07:
Could you send me your dataset? I seem to be unable to reproduce this otherwise

JoseGuzman, 2019-10-07:
Hi Cyrille,

Thank you very much for your interest. Where do you think I could upload the dataset? Is Dropbox a good option for you?

Best

JOse

JoSe GuZmAn • The Guzman Lab
http://www.guzman-lab.com

Tel.:     +43 678 126 70 96 
Email:  jose.guzman@guzman-lab.com

rossant, 2019-10-08:
yes that would be perfect, thanks

JoseGuzman, 2019-10-08:
Hi Cyrille,

[1] the link to the phy files. Please, let me know if you require
other files or further information.

A) You can see that cluster with id=0 and channel=0 has zero amplitudes in
AmplitudeView
B) I also realized that the amp in ClusterView appear 10X larger (they are
correct in AmplitudeView).

I really appreciate your help with these tow issues!. I love Phy!

Best

Jose

[1]
https://www.dropbox.com/sh/wwa4g7zoi14p1zh/AAApvvVxOd6QdjPXW_GKH9U4a?dl=0

rossant, 2019-10-09:
Thanks! could you upload the raw data file too?

JoseGuzman, 2019-10-09:
Sure, I've just uploaded a file called cl_continuous.dat. I think I solved the amplitude issue; the sorting program was not reading properly the gain.

rossant, 2019-10-10:
Fix pushed, is it all working now?

JoseGuzman, 2019-10-10:
Not really, 
1) the reordering in the TraceView is disorganized. This could be due to the sorting program (spyking-circus)
[image]

2) the cluster with id=0, with channel=0 and depth=0 show zeros even though the amplitudes are ~ 20 uVolts.
[image]

JoseGuzman, 2019-10-10:
Yes, 

```bash
(sorting) segundo.martinez@pcl-imba-3:phylib$ git branch 
* master
(sorting) segundo.martinez@pcl-imba-3:phylib$ git log --oneline -1
ff9dbff (HEAD -> master, origin/master, origin/HEAD) alf export test random generator seed for individual test run
(sorting) segundo.martinez@pcl-imba-3:phylib$ cd ../phy
(sorting) segundo.martinez@pcl-imba-3:phy$ git branch
* dev
  master
(sorting) segundo.martinez@pcl-imba-3:phy$ git log --oneline -1
de013cbd (HEAD -> dev, origin/dev) Move FiringRate.x_max attribute from global state to local state

```

I've just used *`--clear-cache`* and *`-clear-state`* the same happens... maybe I need to re-sort again?

rossant, 2019-10-15:
For (1), have a look at your `channel_map.npy` file: phy should take this into account when reordering and displaying the channels in the TraceView.
For (2), try to switch to another type of amplitude by pressing `a`

closed 2019-12-24 by rossant