cortex-lab/phy #895: Error when only one cluster is found.

JoseGuzman, 2019-09-18:
Probably a minor issue, but Phy (latest dev) fails when only one cluster is found. 

`16:29:13.750 [E] __init__:62          An error has occurred (AssertionError): 
Traceback (most recent call last):
  File "/home/segundo.martinez/miniconda3/envs/sorting/bin/phy", line 11, in <module>
    load_entry_point('phy', 'console_scripts', 'phy')()
  File "/home/segundo.martinez/miniconda3/envs/sorting/lib/python3.6/site-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/home/segundo.martinez/miniconda3/envs/sorting/lib/python3.6/site-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/home/segundo.martinez/miniconda3/envs/sorting/lib/python3.6/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/segundo.martinez/miniconda3/envs/sorting/lib/python3.6/site-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/segundo.martinez/miniconda3/envs/sorting/lib/python3.6/site-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/home/segundo.martinez/miniconda3/envs/sorting/lib/python3.6/site-packages/click/decorators.py", line 17, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/segundo.martinez/git/phy/phy/apps/__init__.py", line 164, in cli_template_gui
    template_gui(params_path, **kwargs)
  File "/home/segundo.martinez/git/phy/phy/apps/template/gui.py", line 187, in template_gui
    controller = TemplateController(**get_template_params(params_path), **kwargs)
  File "/home/segundo.martinez/git/phy/phy/apps/base.py", line 364, in __init__
    super(TemplateMixin, self).__init__(*args, **kwargs)
  File "/home/segundo.martinez/git/phy/phy/apps/base.py", line 730, in __init__
    self.model = self._create_model(dir_path=dir_path, **kwargs) if model is None else model
  File "/home/segundo.martinez/git/phy/phy/apps/template/gui.py", line 84, in _create_model
    return TemplateModel(dir_path=dir_path, **kwargs)
  File "/home/segundo.martinez/git/phylib/phylib/io/model.py", line 276, in __init__
    self._load_data()
  File "/home/segundo.martinez/git/phylib/phylib/io/model.py", line 348, in _load_data
    assert self.similar_templates.shape == (self.n_templates, self.n_templates)
AssertionError`

rossant, 2019-12-24:
should be fixed

closed 2019-12-24 by rossant

surygeng, 2022-11-14:
Hi! I still met this problem in 2022. It happens when there is only one cluster. Any advice? 
Here is the error message:
```
15:28:59.135 [E] __init__:62          An error has occurred (AssertionError): 
Traceback (most recent call last):
  File "/home/kang/.local/bin/phy", line 8, in <module>
    sys.exit(phycli())
  File "/usr/lib/python3/dist-packages/click/core.py", line 1128, in __call__
    return self.main(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/click/core.py", line 1053, in main
    rv = self.invoke(ctx)
  File "/usr/lib/python3/dist-packages/click/core.py", line 1659, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/lib/python3/dist-packages/click/core.py", line 1395, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/lib/python3/dist-packages/click/core.py", line 754, in invoke
    return __callback(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/click/decorators.py", line 26, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/kang/.local/lib/python3.10/site-packages/phy/apps/__init__.py", line 159, in cli_template_gui
    template_gui(params_path, **kwargs)
  File "/home/kang/.local/lib/python3.10/site-packages/phy/apps/template/gui.py", line 209, in template_gui
    model = load_model(params_path)
  File "/home/kang/.local/lib/python3.10/site-packages/phylib/io/model.py", line 1348, in load_model
    return TemplateModel(**get_template_params(params_path))
  File "/home/kang/.local/lib/python3.10/site-packages/phylib/io/model.py", line 339, in __init__
    self._load_data()
  File "/home/kang/.local/lib/python3.10/site-packages/phylib/io/model.py", line 434, in _load_data
    assert self.similar_templates.shape == (self.n_templates, self.n_templates)
AssertionError

QWidget: Must construct a QApplication before a QWidget
Aborted (core dumped)

```