cortex-lab/phy #896: Recluster feature in Phy

tanselbaran, 2019-09-18:
Hello, 
First of all, thanks a lot for this great resource! I am using Klusta (3.0.16) for clustering and phy2 kwik-gui for the manual curation part (phy KwikGUI v2.0a1). I noticed the Recluster feature under Edit tab, and thought it may be useful in some cases where there are a few similar clusters that seem incorrectly clustered. However, every time I select the clusters I would like to recluster and press the Recluster button, I first get a "gui:152 Running Klustakwik on N spikes."  message, where N would be the total number of spikes in the selected clusters. And then I immediately get an "actions:191 Error when executing action Recluster" message, without further information. 

Is this an issue that only I am having, or is this feature not fully functional yet? As far as I have seen, there is not much of a mention of it in the readthedocs page either. 
Thanks,
Baran

rossant, 2019-10-07:
could you double check that https://github.com/kwikteam/klustakwik2 is correctly installed and works? If it is, this feature should work!

julienfiorilli, 2019-11-12:
Having the same issue here. @tanselbaran, did you find any solution?

rossant, 2019-12-24:
can't reproduce, please reopen if needed (and post the full console output after running phy with the `--debug` command)

closed 2019-12-24 by rossant