cortex-lab/phy #897: Activate lazo in waveform view

yagui, 2019-09-19:
Hi

This is a feature request.
It would be useful (at least for me :) ) to have the ability to select some waveforms that I would like to remove from the cluster.
Similar to what you can do in the feature view, but in the waveform view.
For example select all spikes that any point of the waveform are inside the selected area.
[image]
In this example I can remove that spike from the amplitude view or feature view, but some times I can not find the way and having this feature will make the cleaning easier.
I don't know how feasible it is to do it. Just a wish :D

Thanks

JoseGuzman, 2019-09-23:
I'm also interested in that. What I did so far, is writing a Phy Script that allows me to move the times of these spikes to a dummy (or 'noise') channel.

yagui, 2019-09-24:
Jose, do you have that code available?

yger, 2019-09-24:
Phy is only showing a fixed number of randomly selected spikes in a given
cluster in the WaveformView. So while the lasso in this view can be useful,
it can not be used to remove manually all "bad" waveforms.

yagui, 2019-09-24:
yger, yes I understand that. But I think the same thing happens in the Feature View, and in the Amplitud View. So you can select certain regions that you might think there is something to remove from the cluster, o select only the region that you want to leave in the cluster, and then do a split.
That is what I do with those views, and having the same feature in the WaveForm View for me it would be a great help.
Also you can manually change the amount of spikes shown in the WaveformView and also I do that to see if there is more noisy Waveforms.
Maybe another feature could it be to toggle between the random selection (as it is now, as I understand), and a probabilistic selection based in likelihood of belonging to the cluster and that will give the opportunity to show a certain amount of outliers.

rossant, 2019-09-24:
When splitting in the feature view, all spikes are loaded before computing the points belonging within the lasso so as to delete all spikes that should be deleted. The problem with the waveform view is that it is currently prohibitively slow to load all spike waveforms before computing the intersection with the lasso, so much as to be unusable in practice.

JoseGuzman, 2019-09-25:
> Jose, do you have that code available?

I do it with the IPython console, I will copy here as a pluging soon.

yagui, 2019-09-25:
> When splitting in the feature view, all spikes are loaded before computing the points belonging within the lasso so as to delete all spikes that should be deleted. The problem with the waveform view [...]

I see. I really don't understand yet how the lasso and the views works.
My idea is to somehow get the points included in the lasso, select only the waveform of the main channel that is being used, and find the ones that are included in the lasso.
For example if a unit has 200000 spikes and assuming 100 data points per waveform, doesn't seem a big amount of data that can't be handled.
Using all the waveforms of all the channels I can see that would be a problem. 
What I mean with the main channel, is the one selected with 'ctrl+left_click' (or the 'bestchannel if no channel has been selected, the same way that amplitudview and featureview seems to work).

I can see there will be an overlap on the lasso command because right now 'ctrl+click' is used for selecting channels, but I think is a minor problem.

rossant, 2019-09-26:
> For example if a unit has 200000 spikes and assuming 100 data points per waveform, doesn't seem a big amount of data that can't be handled.

If a unit has 200 000 spikes, then we need to extract 200 000 waveforms on a single channel. The current waveform extraction procedure is really slow and may take about 1 ms per waveform. So that's about 3 minutes to wait after you've drawn a lasso in the waveform view and requested a waveform split. It's surely not impossible to speed that up, there are several solutions like precomputing the waveforms or considerably improving the waveform extraction code. It just requires quite some work.

yagui, 2019-09-26:
Ok. I see. Thank you for explaining it.

JoseGuzman, 2019-10-07:
> Jose, do you have that code available?

Hi, [here](https://github.com/JoseGuzman/minibrain/blob/master/phy/myClusterView.py) the Plugin I use to remove single spikes (it's very preliminary). The class I use is called SpikeSplitter, and at the moment, I simply use it to prompt in the status bar the id of event that I want to split. After that, I simply type in the IPython console the following command:
`emit('action', s.action_creator, 'split', [spike_ids])`

where spike_ids is a list containing the number of numbers of the spikes that were prompted. At the moment, I do not know how to call emit from the script, that's why I'm doing this.

Let me know if it helps.

Jose

rossant, 2019-10-08:
> At the moment, I do not know how to call _emit_ from the script, that's why I'm doing this.

does this work? `from phy import emit`

JoseGuzman, 2019-10-08:
I tried and it does not work. I think I'm casting properly *spike_ids* in 'emit' .Here is the code

```
import numpy as np

from phy import IPlugin, connect
from phy import emit

class SpikeSplitter(IPlugin):
    """
    This Plugin select single spikes to be splitted 
    """
    def attach_to_controller(self, controller):
        """
        This function is called at initialization time before
        creation of the supervisor object (s) which controls  
        ClusterView and SimilarityView.
        """
        @connect(sender=controller)
        def on_gui_ready(sender, gui):
            """
            Called when the GUI and all objects are fully loaded.
            It makes sure that controller.supervisor is properly defined.
            """
            #------------------------------------------
            # File->Display message
            #------------------------------------------
            gui.file_actions.separator()
            @gui.file_actions.add()
            def show_selected_clusters():
                """
                Display selected clusters in the status bar
                """
                sel = controller.supervisor.selected
                gui.status_message = "Selected cluster/s: %s"%sel
            
            #------------------------------------------
            # in Select->MyPlugins:
            #------------------------------------------
            gui.select_actions.separator()
            myparams = dict(submenu='MyPlugins', shortcut = 'alt+p',
                prompt=True, prompt_default=lambda: 0)
            @gui.select_actions.add( **myparams )
            def split_spike(time):
                """
                split the spike next to the time entered
                """
                #@controller.supervisor.cluster_view.get_ids
                pspike = int(time * 30000)
                # get id from selected cluster
                mycluster_id = controller.supervisor.selected
                # get the sample of the first spike
                t = controller.get_spike_times(mycluster_id[0])
                p = controller.get_spike_ids(mycluster_id[0])
                x = np.argmax(t>time) 
                # remove this p[x]
                #emit('action', s.action_creator, 'split', list(p[x])) #WONT WORK!
                gui.status_message = 'Spike id: %s'%p[x]

```

rossant, 2019-10-09:
How it is not working, do you get any error message, or isn't it doing anything?

JoseGuzman, 2019-10-09:
Yes, I also do not understand it. If I uncomment the line containing emit the **gui.status_message** everything works until that line and phy returns this:

`11:34:14.504 [W] actions:204          Error when executing action Split spike.
`
In $HOME/.phy/config I have

```python
from phy import IPlugin

c = get_config()
# local path to plugins subdirectory
c.Plugins.dirs = [r'/home/segundo.martinez/git/minibrain/phy']

# add the classes to be used TemplateGUI 
c.TemplateGUI.plugins = ['Test', 'ISImedian', 'SpikeSplitter'] 
```

rossant, 2019-10-15:
Actually could you use the following command instead to make a split? 
`controller.supervisor.actions.split([p[x]])`

See this [plugin example](https://phy.readthedocs.io/en/latest/plugins/#creating-a-custom-split-action)

yagui, 2020-07-23:
Hi @rossant 

I started to write a plugin to do this feature and its roughly working. 
https://github.com/yagui/phy-plugins/blob/master/plugins/waveformCuttingView.py
It is a bit slow on big clusters but manageable, 3~4 seconds to load a 50k spikes cluster.

I'm having a few issues than I'm not being able to solve, maybe you can give me some ideas or examples inside phy code, because I was looking the code of all of  the views, but I don't seem to get it.

1) I'm having the same problem that was happening in the feature view, in this issue #857
Could I have some insight where to look?

2) I would like a way to capture the signal emit('select_channel') from the waveformview, as the feature and amplitude views do, but I couldn't make it work.
I tried this snippet of code 
    @connect
    def on_select_channel(self, sender=None, channel_id=None, key=None, button=None):
        print(channel_id)
        print(self)

but although the channel_id is correct, the object self is the waveformview, but not my view, so I don't know how to pass the channel_id to my method 'setChannel'

3) The scaling in the box goes from -1 to 1, with data.bounds (-1,min,1,max). The waveforms  scale to fit the -1 to 1 box, but I would like to have the real amplitude of each waveform. I'm not really sure how to proceed with the scaling and the axes labels. I'm thinking something similar to the amplitude view, but I don't know where to look.

4) I would like to add a key to send a command (move to the next best channel). similar to the 'a' action in the amplitude view
I'm loading all the waveforms of all the channels into one variable. So if I'm able to capture a key, I could switch from channel to channel without needing so solve my question 2 (although I would like to have both)

5) when I close this view, all other views stop working. I have to close phy and restart it

Thanks

yagui, 2020-08-18:
I solved the problem I was having. The solution was to not set any layout.
'Grid' Layout with only 1 subplot, was the closest thing to make it work, because the plots were at the right position, but the lasso was shifted.
'stacked', and 'boxed' not only had the lasso shifted, but also the plots theirself were shitfed.
Based on the ClusterScatterView, I didn't set any layout and it worked perfectly

jiumao2, 2023-02-13:
This is the my plugin to remove strange waveforms: https://github.com/jiumao2/PhyWaveformPlugin
I met the same issue with 'stacked' layout and I did some correction to get the right point position. Here is the code:
``` python
box_size_x, box_size_y = layout.box_size
box, pos = layout.box_map(e.pos)

x = pos[0] * box_size_x * (self.data_bounds[2] - self.data_bounds[0]) / 2 + (
            self.data_bounds[0] + self.data_bounds[2]) / 2
y = pos[1] * box_size_y * (self.data_bounds[3] - self.data_bounds[1]) / (1 + box_size_y) + (
        self.data_bounds[3] - self.data_bounds[1]) / (1 + box_size_y) + self.data_bounds[1]
```